Ultimaker/Cura #9573: Seam relative to object or custom seam per object.

Zogar89, 2021-04-13:
### Is your feature request related to a problem?

Currently we can only select absolute seam or relative seam towards the same place.

This removes flexibility when placing objects on the printing bed and forces the extruder to travel great distances to start each object with the problems that this entails.

[image]

### Describe the solution you'd like

I propose to be able to select the seams independently for each object or perhaps an easier solution is to put a third option, that the seam is relative to each object but does not change direction if we rotate the object, as is currently the case.

[image]

### Describe alternatives you've considered

This is already incorporated by PrusaSlicer with its modifiers so I hope my favorite program (Cura) is not left behind!

### Affected users and/or printers

All

fvrmr, 2021-04-13:
Hi @Zogar89 thank you for your feature request.
You can already do this in Cura with modifiers! 😄 

Select the model and open the modifier menu. Then click on normal model as mesh type and select settings you want to apply for that model.

[image]

Let me know if that works for you!

Asterchades, 2021-04-13:
There's also a checkbox to make the seam relative to the object, not the whole build plate. So instead of being closest to (for example) the centre rear of the plate, it puts it at the rear of the object instead. It won't automatically update on rotating, however, so if you set it to the Rear than it will always be at the Rear - even if that is actually the "Front" of the object after rotation.

Option is called "Z Seam Relative" and only shows when the Z Seam Alignment is set to "User Specified".

Zogar89, 2021-04-15:
Thanks for that, I m really happy.

I generally use the overlay option a lot to modify parts of objects but I had not realized that I could modify entire objects.

I have some considerations to make in the overlay settings tool but it will be in another ticket.

closed 2021-04-15 by Zogar89