Ultimaker/Cura #9576: Start G-Code resets on each app start

DoganM95, 2021-04-13:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Go to Preferences -> Printers and click on Machine Settings for your current printer
2. Delete the text in the automatically initialized Start G-code
3. Close
4. Restart Cura
5. Repeat step 1

### Actual results

Deleting the Start G-code in machine settings is saved for the current session. 
Restarting the application brings back the previously deleted Start G-code.

### Expected results

After deleting the start g-code in machine settings and restarting the application, the start g-code should stay deleted (start g-code should be empty).

DoganM95, 2021-04-13:
A workaround to force the start g-code field to be empty is leaving a comment without any commands:

Currently my start g-code only contains this line:
`; Handled by OctoPrint`

fvrmr, 2021-04-15:
Hi @DoganM95 thank you for your report.
I am not able to reproduce this. I deleted the start g-code for Ender 3 and after restart the start g-code is still empty. Does this happen for all printers?

DoganM95, 2021-04-15:
Hi, 
i only have an Ender 3 in my printers list so far. 

Your question gave me the idea of adding a new printer (now named `Ender 3 #2` in my list) and testing it on that.
Same thing happens to the newly added printer, empty start-gcode field fills up again on Cura restart

fvrmr, 2021-04-16:
Thank you for checking. Could you share your log file, maybe I can find something there.
Can you find your log file here: C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

DoganM95, 2021-04-17:
Sure, i'll attach the file:

[cura.log](https://github.com/Ultimaker/Cura/files/6329779/cura.log)

JohnBaumb, 2021-04-22:
Just curious, @DoganM95, do you have multiple printers defined?

I have two ender 3's and a cr10 s5 defined and it does it unless the workaround is used.
It didn't do it before I added the cr10.

DoganM95, 2021-04-23:
@JohnBaumb no, i only have one printer defined (Ender 3).
Adding profiles does not change anything, the bug persists.

Ghostkeeper, 2021-05-03:
In the log file are 9 starts of Cura. In one of them this line can be found:

    2021-04-15 16:44:55,015 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/plugins/MachineSettingsAction/MachineSettingsPrinterTab.qml:22:38: Unable to assign [undefined] to int

That refers to this bit of code:

    property int propertyStoreIndex: manager ? manager.storeContainerIndex : 1  // definition_changes

So that would mean that `manager.storeContainerIndex` is `[undefined]`. The `storeContainerIndex` property should come from here:

https://github.com/Ultimaker/Cura/blob/7c954fdc2f85eb5caf0131739215b1872e3f69de/plugins/MachineSettingsAction/MachineSettingsAction.py#L53

The problem couldn't be that this function returns the wrong value. If it returned `None` it would appear as `null` in QML, not as `[undefined]`. And we'd also get an error from PyQt that the return value can't be cast to `int`. So the problem is really that this property doesn't exist. Yet `manager` apparently does exist in QML. The question is what this `manager` object then is set to, if it's not an instance of `MachineSettingsAction`. This is not something I could figure out from that log. I can't see how it could be set to something else than a `MachineSettingsAction` since the entire QML is only available if the plug-in is enabled.

DoganM95, 2021-05-07:
@Ghostkeeper i can zip and send you my logged out AppData/Roaming/Cura folder, could this help you reproducing the error?

DoganM95, 2021-05-14:
Just updated my cura to 4.9.1 and all my settings are gone, and account sync did not get my settings back from the servers. 
While updating a dialog popped up asking if i want to delete all my settings... i accidentally clicked enter which confirmed the by default selected "Yes", so no appdata/cura folder left to upload.

Fixing this to a simple **by default unchecked** checkbox asking if existing configs should be deleted while installation is probably more important than the bug i reported initially... The dialog is a terrible solution, i don't understand why it even wants to delete configs..

Ghostkeeper, 2021-05-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/8233.

It's an option in NSIS because some people are adamant that an uninstaller needs to remove all traces of an application when uninstalling. It would indeed be better to make the option more hidden in the uninstaller, if NSIS provides that option to us.

closed 2021-05-17 by Ghostkeeper