Ultimaker/Cura #9578: Ошибки при разглаживании (пожалуйста помогите перевезти)

yankichat, 2021-04-14:
### Application Version

cura arachne

### Platform

Windows 10

### Printer

flyingbear

### Reproduction steps

Я часто использую функцию разглаживание, вот пару багов которые я нашёл:

1- Разглаживание заходит за печатаемую деталь ориентируясь не на дорожки пластика, а на STL модель (при печати большими соплами это заметнее)
[image]

2- Функция границы разглаживания не учитывается 
[image]
[image]

### Actual results

1- Разглаживание загибает крышку или разглаживает в воздухе.
2- Разглаживание всегда происходит до краёв стенок из-за чего появляется волнообразная крышка.

### Expected results

1- Границы разглаживания должны ориентироваться на границы разглаживаемого слоя, а не на границы модели.
2- При увеличении параметра "Границы разглаживания" дорожки разглаживания должны отступать от границ стенок

fvrmr, 2021-04-15:
Hi @yankichat thank you for your bug report.
Could you please fill in the form in English. Thank you in advance.

yankichat, 2021-04-16:
> Привет, @yankichat, спасибо за сообщение об ошибке.
> Не могли бы вы заполнить анкету на английском языке. Заранее спасибо.

I know too little English to write this

yankichat, 2021-04-16:
> Hi @yankichat thank you for your bug report.
> Could you please fill in the form in English. Thank you in advance.

I know too little English to write this

closed 2021-04-16 by yankichat