Ultimaker/Cura #9581: User is confused

Coirt, 2021-04-14:
### Is your feature request related to a problem?

The notification you get when you change settings in a profile and change to a new profile is very confusing for new users.

Please reword the notification box for when you change profiles and have previously changed something from a previous profile.

### Describe the solution you'd like

You have changed setting_(s)_ in the previous profile.
Would you like to **keep** the change_(s)_ and copy them to the new profile?
or **discard** the change_(s)_ and load `this` profiles settings.

2 buttons should be here one labelled "keep" the other "discard"

### Describe alternatives you've considered

The previous profiles setting have been customized.
Would you like to keep the customization and copy them to the new profile?
or discard the customization and load the defaults from 'profileName'

Buttons labelled 
"Keep Changes" and "Load Defaults"

### Affected users and/or printers

This issue raises 2 points. This user thinks that the changes made is either the act of changing profiles or the previous action of making changes to the previous profile. Both issues cause confusion when being asked to discard. Am I discarding the changes made or am I discarding the change I made in the action of changing the profile. Am I keeping the changes I made in the action of changing profiles or am I changing to the changed profiles settings. It really needs to be simplified for new users!

Zogar89, 2021-04-15:
I agree with that!

fvrmr, 2021-04-15:
Hi @Coirt thank you for your feature request.
I will discuss this with the team. Keep you posted!

ksihota, 2021-04-15:
I have also found this dialogue difficult to understand. I assume it is only when switching over (to a new profile) and really makes no changes to the actual profile itself (either the current or the new one), but allows you to keep your current changes with the new profile for this particular slice. Again, I assume that, If you want to retain the changes permanently you would then save the changes to the profile or to a new profile?
In my case I would prefer that it reminded me that I have made changes that may need to be saved to my existing profile before switching to a new one. It could then ask me if I would like these current settings to be retained and used with my new profile.
Some clearer vocabulary or possible some more in depth explanation of this dialog would certainly be helpful.

fvrmr, 2021-04-16:
I have discussed this with the team and our UX designer is going to take a look at this. Thank you all for mentioning that this seems to be unclear!

GregValiant, 2021-04-16:
My suggestion for the text at the top of the "Discard or Keep changes" dialog box:

"You have **over-ridden** some of the profile settings of "{profile_name}"."
"You can KEEP those changes as temporary over-rides to "{next_profile}" or you may DISCARD the changes and "{next_profile}" will be loaded with it's default settings."
"(The current profile {profile_name} will be unloaded with no changes to it's defaults.)"

Maybe change the "discard" button text to "Discard Changes and Load Defaults"

Coirt, 2021-04-16:
> I have discussed this with the team and our UX designer is going to take a look at this. Thank you all for mentioning that this seems to be unclear!

Thank you!

Ghostkeeper, 2021-05-04:
Developers, see issue CURA-8210.