Ultimaker/Cura #9582: Cura slices parts I deliberately float above the build plate like they are ON the build-plate...

Chromatamata, 2021-04-15:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Creality CR-10 mini

### Reproduction steps

I have recently started to use a work flow of designing 3D objects that are prepared in such a way as to create multicolour prints using a single nozzle printer. The slicing technique to generate such objects requires that some parts be sliced in mid-air above the build plate and not in contact with it.

I turn off 'Automatically Drop Models to the Build Plate'. I import my models and position them correctly relative to each other using the 'merge models' function and the 'move' tool, situating the lowest model ON the build plate. I then, ungroup the models and delete all but one model and slice it and save my file. I undo all the deletes and re-delete models until I have another single model ready for slicing in Cura. Then I slice that model and save the file. I continue cycling through until all aligned models have been separately sliced and saved.

I am having two difficulties as outline in the actual results section.

### Actual results

1. If you examine the gcode, despite my NOT having 'Automatically Drop models to the build plate' tuned on, Cura slices all parts that are visually floating above the build plate as though they are sat flush with it anyway!!! I don't know why I can turn the setting 'Automatically Drop models to the build plate' on and off or what use it is other than being visually pleasing, if all parts are sliced as though they are sat on the build plate regardless of this setting...

2. Parts which are positioned, floating more than a certain height above the build plate will not slice AT ALL!!!!!!!!!!!! I cannot seem to replicate this second problem though as it seems to be random in nature.

Please find an attached project file and resultant gcode in zip file. Notice how the initial z-height is set to the layer height and not the height of the model above the build plate plus layer height.

[Cura slices everything as though it is on the build plate.zip](https://github.com/Ultimaker/Cura/files/6316410/Cura.slices.everything.as.though.it.is.on.the.build.plate.zip)

### Expected results

I expect and in fact NEED all parts to be sliced with the true z-height of where they are (even, in fact, ESPECIALLY if they are floating) not lowered to the build plate in the gcode.

fieldOfView, 2021-04-15:
See the setting `Remove Empty First Layers` (in the `Mesh Fixes` category)

Chromatamata, 2021-04-15:
Ok. Not a bug. My apologies. I have unticked that box and it now works as expected. It was recommended I report this by a moderator on the Cura forum where no one had a solution. Might I suggest that the ticking and unticking of this property be merged with the ticking and unticking of 'Automatically Drop Models to the Build Plate' as this particular repair setting is setting is not something I have ever played with and I was not aware of its existence. I could be wrong but I can't see a reason for the two settings to operate independently. If you have unticked 'Automatically Drop Models to the Build Plate' then you probably want to slice models away from the build plate. Just a suggestion...

fieldOfView, 2021-04-16:
`Automatically Drop Models to the Build Plate` and `Remove Empty First Layers` are two very different things. They may be similar in your workflow, but not in mine.

Chromatamata, 2021-04-19:
I guess they are different but they only differ in where the model is 'cut' for slicing and only then when the model is situated at a height above the build plate such that z<>multiples of the layer height. For every z that is an exact multiple of the layer height, the resultant gcode will be exactly the same, regardless of whether the model was first lowered to the build plate or raised to some height that is a multiple of the layer height above it. It's hard to find that setting, regardless of whether they differ or not. Perhaps unticking Automatically Drop Models to the Build Plate could have a dialogue box which asks the user if they also wish to untick Remove Empty First Layers? It would help the user understand that turning off Automatically Drop Models to the Build Plate isn't enough to generate Gcode above the height of the build plate.

Chromatamata, 2021-04-19:
Ok. Further to this, although I became convinced that this wasn't a bug as when I first sliced a model after unticking the box 'Remove Empty First Layers' and examining the gcode, everything appeared to work and the the print nozzle was being sent to the expected height, I am going to have to re-instate it as a bug. Having the box unticked NO LONGER produces the correct results. I don't know why. Does Cura perhaps read the check box as ticked when it is in fact unticked? The box is still unticked on the profiles I am slicing with and yet Cura has returned to its old behaviour of erroneously generating gcode for the models as they they are level with the print bed.

Chromatamata, 2021-04-19:
Ok. I have done further testing and it appears to be completely random. For example, sometime slicing a model ABOVE the build plate generates gcode ON the build plate, sometimes it generates gcode at the intended height above the build plate and sometimes it produces gcode at a seemingly random height above the build plate. By way of example, if I slice the cube in my example at 0.4mm above the build plate with 'Remove Empty First Layers' turned off the returned gcode has the model sitting at 0.8mm above the build plate. At this stage, I am baffled. This is very strange behaviour.

Ghostkeeper, 2021-05-04:
> I could be wrong but I can't see a reason for the two settings to operate independently.

The slicing setting Remove Empty First Layers should really always be enabled regardless of what the user is printing. The only raison d'être for this setting is really to resume previously cancelled prints or otherwise print stuff on top of existing items in the build volume (like maybe textile printing). Basically the poor man's Z offset. On the other hand dropping items to the build plate is quite commonly disabled when people have multiple objects in the scene. If these objects are aligned in certain ways you wouldn't want to drop them down, especially things like infill meshes.

Basically, Remove Empty First Layers applies globally to the entire print, while dropping models to the build plate is individual to the model. Removing Empty First Layers also guards against parts being too small to print.

> By way of example, if I slice the cube in my example at 0.4mm above the build plate with 'Remove Empty First Layers' turned off the returned gcode has the model sitting at 0.8mm above the build plate. At this stage, I am baffled. This is very strange behaviour.

Okay, so I'm trying these steps:

1. Load `1_cm_Cube_(Meshed) 1.3mf` from the first post of this thread.
2. Disable Remove Empty First Layers.
3. In the Move tool, set Z to 0.4.
4. Slice and save the g-code to disk.

I'm then getting the following snippet of g-code:

```
<Whole bunch of start g-code>
;
; END 'My Start G-code' *********
;
;
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:52
;LAYER:2
M106 S170
;MESH:1 cm Cube (Meshed).stl
G0 F12000 X145.6 Y114.4 Z0.6
;TYPE:WALL-INNER
;BRIDGE
M106 S255
G1 F2700 E0
G1 F1200 X145.6 Y105.6 E0.14634
```

So it's moving to Z0.6 just before starting printing. Since the Initial Layer Height is set to 0.2mm, the top of the first layer should be at 0.4mm + 0.2mm = 0.6mm, so Z0.6 is correct.

Are there any other reproduction steps I should've taken into account to test the bug you found? I've tried a few things, sliced multiple times, with support and without, etc. but it seems to be working fine.

Chromatamata, 2021-05-04:
> The only raison d'être for this setting is really to resume previously cancelled prints or otherwise print stuff on top of existing items in the build volume (like maybe textile printing)

The 'raison' in my case is for the purposes of using a multicoloured printing technique with a regular single nozzle printer that doesn't seem to be terribly well-known or used. I figure we'd see a lot more of it if more people knew about it because it is an AMAZING way to print spectacular objects.

> Are there any other reproduction steps I should've taken into account to test the bug you found? I've tried a few things, sliced multiple times, with support and without, etc. but it seems to be working fine.

I have no idea, unfortunately. :( All I know is that it sometimes works correctly for me but doesn't most of the time and it seems very random. I am using Cura 4.8 and have not yet updated to 4.9. maybe that's why. 

I have basically given up relying on Cura to spit out the correct z-height results and have instead got around my problem by insisting Cura remove the empty layers (which is what I don't want but that way I can keep it under control) and have written my own g-code post-processing program to take Cura's output and correct it while also joining the multiple parts into one piece of g-code that serves my purposes. I am happy with this solution at present and it is working for me to achieve my goals. Others may not be happy with this solution or able to do this for themselves.

I am baffled and suspect there IS actually something wrong with Cura's code somewhere but I can't make it do it wrong on command. If I have any more insight into this I will let you know.

nallath, 2021-05-05:
> The 'raison' in my case is for the purposes of using a multicoloured printing technique with a regular single nozzle printer that doesn't seem to be terribly well-known or used. I figure we'd see a [...]

Do you have an example of that? I know that quite a few people use the pause post-processing script to get something similar to what you are describing.

Chromatamata, 2021-05-05:
> I know that quite a few people use the pause post-processing script to get something similar to what you are describing.

No, you cannot just use the pause post-processing scrip to achieve what I'm doing. All that will do for you is to make stripy prints. With this technique, you can can create prints that display pictures and text in anywhere from 2 colours to full colour images or at least as many colours as you have different colours of filament. Theoretically, you could also use some of your filaments as primaries to create half-toned images although I've never tried it and am not sure how well it would work. The process requires models to be broken into several, sometimes many, parts and then sliced separately (but together) and the resultant output recombined into usable gcode. The various colours, however, are all on the same print layer. With it, you can put full colour images on the either the lower or upper surfaces of prints or if combined with 2D nets on all surfaces of a print. 

> Do you have an example of that?

I am personally only just beginning to explore and hone this technique so I don't have a great deal of examples to show you and nothing in spectacular full colour yet. I have been held back in my progress by having to overcome, manage and work around these strange unexpected behaviours in Cura which, if I couldn't program for myself, would have prevented me making this kind of print. However, I have attached a picture of two objects I recently designed and prepared in a CAD program and printed for my little niece. I hope to have some even more interesting prints soon.

[image]

Ghostkeeper, 2021-05-06:
The Z Offset plug-in from the Marketplace may provide a suitable workaround then.

But we're going to need reproduction steps then that reproduce this problem. So far we haven't seen that bug yet.

closed 2021-05-20 by no-response[bot]