Ultimaker/Cura #9583: [4.9 beta] Make overhang printable affects cutting meshes

printingotb, 2021-04-15:
### Application Version

4.9 beta

### Platform

win 7

### Printer

custom fff

### Reproduction steps

Enable "make overhangs printable", I used 33°.
Place a cutting mesh with for example reduced speed.

### Actual results

Cutting mesh is affected by make overhangs printable.

### Expected results

It shouldn't be affected?
[image]
[overhang printable.zip](https://github.com/Ultimaker/Cura/files/6316827/overhang.printable.zip)

fvrmr, 2021-04-15:
Hi @printingotb thank you for your bug report!
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-04-15:
It might sound a bit weird going by the opposing titles, but I think this has the same root cause as this bug: https://github.com/Ultimaker/Cura/issues/9350

The bug is that instead of "Make Overhang Printable" being applied to some part of the main mesh, it's being applied to the cutting mesh, causing a different piece to be cut off. Your bug report reports that a different piece is being cut off. The other one reports that (part of) the main mesh is not being edited.

printingotb, 2021-04-15:
Sounds like the same bug.

If I don't disable "make overhangs printable" with the modifier, the modifier is affected by the settings profile.
[image]
If I do disable it, its affected to a certain amount, the model is still affected:
[image]
With the setting turned off in the profile and then enabled in the modifier, neither is affected.
[image]
[Body1.zip](https://github.com/Ultimaker/Cura/files/6320745/Body1.zip)
(sliced in arachne beta this time)

fvrmr, 2021-04-16:
Hi @printingotb I have discussed it with the team. And we decided to defer this issue. Because we think that not many people will use this combination of settings. Hope you understand.

printingotb, 2023-05-27:
Confirming that the issue is still present in 5.3.1

MariMakes, 2023-05-30:
Hey @printingotb,

Fresh eyes reporting for duty 🐣 

Thanks for doublechecking if it's still present. It's odd indeed. 
I did notice that you can disable the setting in your mesh, but I wouldn't have thought of that when I saw it. 
[image]

I'll bring it up to the team to see what they think. 
But I expect that if it's not a quick fix that it will not get a high priority to implement.

printingotb, 2023-05-30:
Thanks, keep up the good work!

MariMakes, 2023-06-01:
I brought your issue up with the team. 💪 
They recognized the issue but since there is an easy workaround in place, and the setting is experimental we decided to close the issue. Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I'll close this issue, I hope you understand. 🙏

closed 2023-06-01 by MariMakes