Ultimaker/Cura #9584: Modification/New Feature -  Multiple "Initial Layers"

KenLucke, 2021-04-15:
### Is your feature request related to a problem?

I prefer to print my first 3 layers or so with a different temperature for bed and filament, which helps cut down on "elephant's foot" on my finished models.  Currently, Cura only allows you to set a different temp for the filament or bed on the first layer only.  This forces me to either manually change the temps when the print gets to the appropriate layer, or mess around using post-processing scripts to change things.

### Describe the solution you'd like

I'd like to see the "Printing Temperature Initial Layer" setting for filament changed to allow a set NUMBER of layers for which the setting applies, and [even better] an optional number of "fade" layers to gradually change to the temp for the rest of the print (kind of like Marlin's Z-fade for bed leveling gradually adjusts back to no adjustments).  Also, the same setup for the "Build Plate Temperature Initial Layer" settings.

### Describe alternatives you've considered

Live with it the way it is:)

### Affected users and/or printers

Many

fvrmr, 2021-04-16:
Hi @KenLucke thank you for your feature request. I wil discuss this with the team.
Keep you posted!

Ghostkeeper, 2021-04-21:
We've decided to defer this feature request. We don't consider it a problem for the build plate since the build plate often takes a while to cool down anyway. There is more to gain from the nozzle temperature but there we already have a similar feature: The fan speed can be reduced for more than just the initial layer (see the Regular Fan Speed At Height/Layer setting).

I wouldn't be surprised if it comes up from within Ultimaker as well later on though. But we won't be working on this any time soon.

closed 2023-07-13 by github-actions[bot]