Ultimaker/Cura #9585: Travel Options

Ivon365, 2021-04-16:
### Is your feature request related to a problem?

I make whistles with my 3d printer and I'm frustrated when the toolpath (extruder path) doesn't stay within the infill (already printed parts) when traveling to the next spot to print. The travel lines show that it has to go over the whole print to get to one spot.

### Describe the solution you'd like

Keep the travel within printed areas if at all possible (even if it means going over areas already printed). Just make sure to keep the filament retracted until it gets to the area that still needs printed.

### Describe alternatives you've considered

I've already tried the "Infill Travel Optimization" option within Cura and it still doesn't travel within the infill.

### Affected users and/or printers

This will affect all Ender 3/ Ender 3 Pro printers as well as custom FFF printers using Marlin.

Asterchades, 2021-04-16:
Under Travel, look for Combing Mode. Set it to "Within Infill" or "Not in Skin". If you still want it to retract during those movements, set a small "Max Comb Distance with No Retract" value.

fvrmr, 2021-04-16:
Hi @Ivon365 thank you for your feature request. And like @Asterchades mentioned above try the combing mode settings.
Let us know if that works for you.

closed 2021-04-30 by no-response[bot]