Ultimaker/Cura #9588: 3mf project files don't get a model check for errors

GregValiant, 2021-04-16:
### Application Version

4.8.0   4.9beta

### Platform

Win 10

### Printer

Creality E3Pro

### Reproduction steps

1. Open a 3mf file that includes a model that has errors. 

### Actual results

 The "model is not watertight" warning doesn't come up.
I opened the attached 3mf file.  There was no warning.
Just to check, I exported the model from Cura as a binary STL and then cleared the build plate and brought the STL in to Cura and the warning did pop up.  This is a model from a poster on the Ultimaker Forum.
[E3PROSKR_eagle-Body3_LW-profile3_PPPLWPLA-0-00-025.zip](https://github.com/Ultimaker/Cura/files/6326089/E3PROSKR_eagle-Body3_LW-profile3_PPPLWPLA-0-00-025.zip)
 

### Expected results

 The "model is not watertight" warning will come up as it does when an STL file with errors is opened.

fvrmr, 2021-04-19:
Hi @GregValiant thank you for your bug report.
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-04-29:
I couldn't reproduce this bug. I tried it with a 3MF file of my own first:
[image]

And also with the project file in this ticket:
[image]

But I think I know what the issue is here. The frequency at which the check for model errors occurs was significantly reduced in Cura 4.8, for performance reasons: https://github.com/Ultimaker/Cura/commit/333b2959cd8c43bd580226bb5dfe0fb17587cf93
So it takes up to 30 seconds for the message to appear. That is really quite a lot! Should we maybe also trigger a check just after the file is loaded, to catch 90% of the cases within a fraction of a second?

GregValiant, 2024-11-19:
This is still an issue in 5.9.0-beta.2.
To add a little bit:
If I open a project file and "import models" then the model is checked for errors.
If I open the same file as a "project" then there does not appear to be a model check.