Ultimaker/Cura #9592: Crash on editing materials - Print Setting Tab

terryrobison, 2021-04-17:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Custom Printer 

### Reproduction steps

Open Materials, select a material.  The "Information" tab can be edited.  Selecting the "Print setting" Tab results in a fatal error.    The program crashes with a Crash Report window open.

Error traceback:
Traceback (most recent call last):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Settings\Models\ContainerPropertyProvider.py", line 84, in setKey
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Settings\Models\ContainerPropertyProvider.py", line 148, in _update
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Settings\Models\ContainerPropertyProvider.py", line 167, in _getPropertyValue
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Settings\SettingDefinition.py", line 617, in settingValueToString
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Settings\SettingDefinition.py", line 755, in <lambda>
TypeError: float() argument must be a string or a number, not 'SettingFunction'

Logs:
Thread 0x00001084 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005cb4 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005138 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000043f4 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000405c (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000679c (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005c90 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004640 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000040c4 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002368 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x000005c4 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

### Actual results

The program will always crash at this point.    

### Expected results

Selecting the "Print settings" in the Materials profile shouldn't crash the program

fvrmr, 2021-04-19:
Hi @terryrobison thank you for your bug report.
Looking at your log I assuming that you mean 4.9 instead of 4.8?
Do you maybe have an X driver on your computer?

fieldOfView, 2021-04-20:
Please post a link to the entire Cura.log and not just the error traceback.
Do you have any plugins installed?

terryrobison, 2021-04-20:
Hi,
I get the same behaviour in both 4.9. and 4.8.   No plugins installed with
4.9 / and with 4.8 there are plugins installed.  I have removed and
restalled both versions removing everything in the APPDATA area before
reinstalling.

Regards,

-- 
Terry

"You can log off / But you can never leave."
 --------------------------------------------------------------
 -  Terry Robison
 -  Senior Systems Analyst
 -  Blueprint I.T. Services Pty. Ltd.
 -  ***@***.***         _\|/_
 -                                                   (o o)
 -------------------------------------oOO-(_)-OOo----

fieldOfView, 2021-04-20:
Cura.log can be found via Help -> Show configuration folder. Please upload that file to google drive, onedrive, dropbox, or the like, and post a link here.

closed 2021-05-04 by no-response[bot]