Ultimaker/Cura #9593: Cura Ironing below final layer causing future layers to not print in specific circumstances

Altirix, 2021-04-17:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Modified Ender 3

### Reproduction steps

Im using Cura 4.8.0 with the profile below, and i am slicing the model attached (i am also using support blockers)

once sliced im sending the gcode to Octoprint+Klipper which is also running ArcWelder

it could be any of these programs in the chain or even just my configuration (let me know if you need those to figure out what is causing this) but im suspecting the most probable cause is cura and how ironing is done as this fails at the exact layer finishes ironing a surface that is below the top of the model

### Actual results

once the printer gets to layer 134 or 135 the printer stops extruding the next layers, the extruder stepper is still moving however after a few layers filament comes out again. this is not a clog, the same failure occurred twice on this model, hotend is still at correct temp.

### Expected results

hotend continues to extrude and make next layers

Altirix, 2021-04-17:
[bugreport.zip](https://github.com/Ultimaker/Cura/files/6328573/bugreport.zip)

Contents 

Cura Profile
sliced Gcode
sliced Gcode + ArcWelder

I lack the STL with the support blockers and model but the model can be downloaded here

https://www.thingiverse.com/thing:4753906

fvrmr, 2021-04-19:
Hi @Altirix thank you for your bug report.
A project file (file - save project) would be nice to share. I can't reproduce this how you described above.

closed 2021-05-03 by no-response[bot]

Ghostkeeper, 2021-05-04:
> once the printer gets to layer 134 or 135 the printer stops extruding the next layers, the extruder stepper is still moving however after a few layers filament comes out again. this is not a clog, the same failure occurred twice on this model, hotend is still at correct temp.

Cura assumes that if it outputs E commands, the feeder motor will be rotating, and that if the feeder motor is rotating, it extrudes or retracts material. If you're observing the feeder motor rotating but not extruding, then you may be experiencing slip in the feeder. It would be good to reduce the aggressiveness of retractions then.

If you can point out something that Cura does wrong, I'd love to help. It could be that Cura instructs the printer to retract but then not to unretract or something. However with the given reproduction steps I only get 131 layers at all in the print, using the "orbiter back" model and your profile. Layers 134 and 135 don't exist.