Ultimaker/Cura #9594: After 3-5 clicks cura allways freezes for 30-40 seconds then you can work again

awiltschek, 2021-04-17:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Sidewinder X1

### Reproduction steps

I have allready uninstalled all possible extensions (including the octoprint) but there was no change in this behavior and when looking into the logfiles it seams that cura allways wants to communicate but stucks with it.
its totaly cumbersome and working with cura gets more and more miserable.
so i load a model - cura stuck - after 30 seconds you can continue working
you make some settings and slide - cura stuck - after 30 seconds you can continue
and this the whole time
i uninstalled cura and installed fresh one 
i uninstalled ALL unnecessary extensions - no changes
yes i can install a new windows - you dream it

### Actual results

see above

### Expected results

working without any "program doesnt react" delays

fieldOfView, 2021-04-18:
Do you perhaps have a network drive mapped to the drive-letter X? If so, please try mapping that to a different derive-letter.

awiltschek, 2021-04-18:
no there is no network drive mapped at all.

fieldOfView, 2021-04-18:
Just to troubleshoot, could you try disconnecting all network connections and then try Cura again?

awiltschek, 2021-05-02:
Sorrry for late reply. Yes without network connection its working smooth. So what shall i do now disconnect everytime my network when working with cura ?

fieldOfView, 2021-05-02:
You could try going to the "Installed" tab of the Marketplace, and uncheck the "Ultimaker Network Connection" plugin. Then restart Cura and see if the slowdown still occurs.

awiltschek, 2021-05-02:
i do not even have this option in my cura 4.8 - but i now disabled the firmware checker - he might also connect to network

awiltschek, 2021-05-02:
no improvement with the firmware checker disabled
still searching for your network conncetion setting

fieldOfView, 2021-05-02:
Sorry, I meant "UM3 Network Printing". Could you have a look again?

awiltschek, 2021-05-02:
this option is disabled but no improvement
any other idea ?

awiltschek, 2021-05-02:
this was allread disabled. any other ideas ?
yes when pulling network cable cura works well.
when i have network connection it has the issue described.

fieldOfView, 2021-05-02:
The "UM3 Network Printing" plugin is the only default plugin that continuously responds to certain network traffic. If that was already disabled, and you don't have other plugins like the OctoPrint Connection plugin or similar plugins that connect to printers over the network installed, then I don't know what could be causing the slowdowns.

Please upload your cura.log somewhere and post a link here.

awiltschek, 2021-05-03:
[cura.log](https://github.com/Ultimaker/Cura/files/6414548/cura.log)

Ghostkeeper, 2021-05-05:
One possibility is the autosave. If there's like a million profiles to save, maybe it could take some significant amount of time. In your log file there is usually between a few seconds and a few minutes of inactivity after the autosave. But that is the nature of the autosave in general; it's supposed to trigger after 10 seconds of inactivity so it'll often happen when you're not doing anything that would create log entries.

Could be that the AppData folder is somehow slow to write to? According to the log file it's fast enough to read from though; you're not experiencing significant loading time during start-up.

We haven't been able to reproduce this issue yet, so far.

nallath, 2021-05-06:
I've added a bit more logging to the autosaving to see if that's the culprit.

awiltschek, 2021-05-06:
I just have 8 profiles which i save on my own. or what kind of autosave do you talk about. 
so far my ssd didnt show performance lacks and however saving local files is influencing the performance of the network is not clear for me, as i allready stated that pulling the network cable helps cura to perform accordingly.

awiltschek, 2021-05-06:
i now tried to install Cura 4.9.0 and i got this screen on starting it - what do you do ?
[image]

nallath, 2021-05-06:
> i now tried to install Cura 4.9.0 and i got this screen on starting it - what do you do ?
> [image]

That is a different issue and one that should be fixed for the 4.9.1 release that is expected to be available next week.

awiltschek, 2021-05-06:
maybe you should then take this version from the download are until v4.9.1 is available ?

fieldOfView, 2021-05-06:
There are many more people for whom it works than for whom it does not work.

nallath, 2021-05-06:
> maybe you should then take this version from the download are until v4.9.1 is available ?

The issues are not to that extent that we feel that this is needed / required.

awiltschek, 2021-05-06:
is there now any proposal for my issue ?

fieldOfView, 2021-05-06:
The hypothesis that Autosave is the problem can be tested by changing the autosave interval.

* Go to Help -> Show configuration folder.
* Close Cura
* In the configuration folder, locate `cura.cfg` and open it in a text editor (eg notepad)
* Locate the `[cura]` section and add the following line to it:
`autosave_delay = 3600000`

awiltschek, 2021-05-08:
ok i found now the issue and maybe you should make a recommendation for others.

You must not store your STL Data on Microsoft OneDrive during slicing.

I moved now the STL (Inputfiles) onto my local disc and a wonder happended, Cura has no timeouts any more.

So its not autosave - just the bloody Ms Onedrive which has on my machines certain outages (maybe because of unnecessary syncs or autosaves. althout my internet performance is quite hight 75MBit).

Sorry for bothering you for this microsoft mess !
Thnx for your help

fieldOfView, 2021-05-09:
Sounds like it could be related to the detecting updates in the file then. Perhaps there should be an option to turn that off.

nallath, 2021-05-10:
> you mean in onedrive an option or in cura.
> currently onedrive and the micrsosoft mess is getting worser from update 2 update.
> multiple delays when working on a slide or in excel.
> so microsoft is realy doing a bad job again.

Within Cura. Cura attempts to check if the file was updated since it last loaded it in order to present he user with a "File has changes on the disk, do you want to reload" popup.

Ghostkeeper, 2021-05-17:
Perhaps a nicer solution would be to make sure that watching for file changes doesn't take the main thread or the GIL.

Vandresc, 2021-05-18:
Devs see CURA-8243 for our internal reference

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

awiltschek, 2024-11-19:
yes - as i dont use cura anymore - you can close this.

closed 2024-11-19 by GregValiant