Ultimaker/Cura #9595: Having blank start/stop gcode regenerates default start/stop gcode on restart.

JohnBaumb, 2021-04-17:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3 / CR10 S5

### Reproduction steps

Open Cura
Check printer start/end gcode, delete (using klipper, don't need it).
Restart Cura
default gcode reappears.

May be related to having multiple printers defined (I have 3 defined, two ender 3's and a CR10 S5).

Able to mitigate this issue by simply adding a semicolon for each category.

### Actual results

Since there is no notification, you can export sliced gcode and it will work correctly, but the next time you use cura, your gcode will contain the default gcode. Thereby creating undesired results without any notifications to the user that something has changed.

### Expected results

Blank gcode should stay blank, perhaps have a button for reverting default gcode if necessary.

GregValiant, 2021-04-17:
As a work around, have you tried leaving 1 comment in?
**;**Klipper

JohnBaumb, 2021-04-17:
Yes, I mentioned that in my submission, and it appears to work.

fvrmr, 2021-04-19:
Hi @JohnBaumb thank you for your bug report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9576 that's why I will close this issue. 
The work around that @GregValiant is suggesting works. But I am not able to reproduce this issue unfortunately.

closed 2021-04-19 by fvrmr