Ultimaker/Cura #9597: top/bottom section settings menu not showing properly with custom settings visibility plugin

brunoosti, 2021-04-17:
Top/bottom section settings menu only shows in the pre defined visibilities options. 
With custom settings visibility plugin, even after selecting top/bottom stuff in the preferences window, the Top/bottom section does not show in the UI.
Also had problems after selecting a pre defined visibility option (Expert, lets say) with "Top/Bottom thickness" setting not showing, only its separated sub settings "top thickness" and "bottom thickness". Closing and collapsing the section again seamed to make all the setting reappear. 

[image]

fieldOfView, 2021-04-18:
The Setting Visibility Presets plugin is a thirdparty plugin (created by me). Please report issues with it here:
https://github.com/fieldOfView/Cura-SettingVisibilityProfileCreator/issues
Does this issue of the new category not showing up happen with Setting Visibility Sets you created in Cura 4.8 which carried over to Cura 4.9, or with Setting Visibility Sets you created in Cura 4.9 (or both)?

Does the second issue also happen without the plugin installed? Please report one problem per issue.

jbeima, 2021-04-18:
I am also seeing this...

fieldOfView, 2021-04-18:
> I am also seeing this...

Could you please be more specific than "this"? I really don't have enough information to know what the issue is.

jbeima, 2021-04-18:
When you look at your profile and you go to make changes...

In 4.8 there was "Shell". It looks like in 4.9 & the new engine this has been split into two sections. "Walls" & "Top/Bottom".

The "Walls" sections appears just fine and you can go into it and make any changes as expected. However the "Top/Bottom" grouping does not appear and any settings in it can no longer be edited. Does not matter if you go into the settings and uncheck everything under it and then re-check them. That "group" will simply not appear in Cura.

NOW. I see this in tha Aracne Engine Beta but NOT in the 4.9 Beta. For me 4.9 works as expected.

I am running a machine that I have had 4.8 on for a long time, so it did migrate all my existing printer/setting profiles. Not sure if that has something to do with it or not.

In 4.9 the profiles were migrated, but all of the visibilty on/off checks were LOST. The "Top/Bottom" group appears as expected.

In Arache the profiles were migrated and all settings were maintained. The "Top/Bottom" group does NOT appear and no matter what I do I can not make it appear. Even if I copy a profile and edit the new one, still nothing.

brunoosti, 2021-04-18:
It happens only with setting visibility saved sets from older cura version (came from 4.8 to Arachne in my case). I never tested without the plugin. But this issue seams to come from the new way menus are organized for the new Arachne engine (from "she'll" to "walls" and "top and bottom"). 
jbeima described the issue very well. 

The second proble m- top/bottom stuff not showing after selecting a default visibility set - must have had something to do with selecting from one of the plugin's saved sets, because it didn't happen again. 

The plugin must be having problems identifying the new divided categories "walls" and "top and bottom", maybe.

Answering from my phone. Once I get to the computer I'll copy this to the plugin github page. Or should we continue here?

jbeima, 2021-04-19:
If I had to guess, it has to do with the grouping. The migrated profile has all the "settings" associated to the Shell group and either only migrated some of them to Walls, or Walls just displays what was in Shell. Since the settings exists in the "Shell/Wall" group they do no even allow for Top/Bottom to show up, and nothing changes the group the settings belonged to, to the new group. Since they are not in the new "Walls" group, they do not display there. Effectively leaving them undisplayed anywehere.

fieldOfView, 2021-04-19:
> Once I get to the computer I'll copy this to the plugin github page. Or should we continue here?

The plugin only creates setting visibility set files. From what you describe, these files don't get properly upgraded when migrating the settings from Cura 4.8 to Cura 4.9. That sounds like a bug in the way Cura 4.9 upgrades the setting visibility sets. So for now let's keep this here, unless the Cura devs say otherwise.

fieldOfView, 2021-04-19:
Wait, rereading what both of you said, it starts to make sense; Cura 4.9 is behaving properly. 

I think this is a known issue affecting the Arachne builds only. The Arachne builds don't migrate/update the setting visibility sets, but the 4.9 beta does.

brunoosti, 2021-04-19:
Let me edit the title so there is no more confusion. From the beginning I'm referring to cura Arachne beta. Not sure why, but I Wrongly though this issues section was only for that version

fvrmr, 2021-04-19:
Hi @brunoosti thank you for your bug report. 
This was already reported before on the discussion for Arachne beta. https://github.com/Ultimaker/Cura/discussions/9388#discussioncomment-480207

I will discuss this with the team. Keep you posted.

brunoosti, 2021-04-19:
Oh! Sorry! I've searched for this but couldn't find anything. I'll close it then

closed 2021-04-19 by brunoosti

Ghostkeeper, 2021-04-21:
Indeed this was a bug that was fixed in 4.9 though a [version upgrade](https://github.com/Ultimaker/Cura/blob/b21c9a3b24d3ab9efb4c082768d28a4314488845/plugins/VersionUpgrade/VersionUpgrade48to49/VersionUpgrade48to49.py#L32). It is also fixed in Arachne now (but after the Arachne beta was released).

jellespijker, 2021-04-21:
A work around might be to first load the project in Cura 4.9 (one time only) and the. Open them in Arachne beta. Opening them in 4.9 should take care of upgrading the visibility categories.