Ultimaker/Cura #9600: Failure to launch:

tormenteddave, 2021-04-18:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

none

### Reproduction steps

1. Install Cura 4.9.0 Beta
2. Launch Cura
3. Get error message.

### Actual results

cx_Freeze: Python error in main script (traceback unavailable)

error in sys.excepthook.
Exception: [WinError 43] The network path was not found: 'x:\\4.9-exe\\build\\inst\\bin\\cura_app.py'
Original exception: [WinError 53] The network path was not found: 'x:\\4.9=exe\\build\\inst\\bin\\cura_app.py'

[image]

### Expected results

Click 'ok' and it ends the program.  It never launches the app.

tormenteddave, 2021-04-18:
running it with Admin rights let me in to the program.

fvrmr, 2021-04-19:
Hi @tormenteddave thank you for your bug report.
Could you share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

tormenteddave, 2021-04-19:
[cura.log](https://github.com/Ultimaker/Cura/files/6336415/cura.log)

Cobra813, 2021-04-24:
> ### Application Version
> 4.9.0
>
> ### Platform
> Windows 10
>
> ### Printer
> none
>
> ### Reproduction steps
> 1. Install Cura 4.9.0 Beta
> 2. Launch Cura
> 3. Get error message.
>
> ### Actual results
> cx_Freeze: Python [...]

Got the Same issue.

evanthomas, 2021-04-24:
Me too - identical.

fieldOfView, 2021-04-24:
If you have a drive mapped to the driveletter X, please use a different drive letter as a workaround.

Ghostkeeper, 2021-05-05:
This will be fixed with Cura 4.9.1, by moving our build folder on the Windows build server to the `C:` drive rather than the `X:` drive.

closed 2021-05-07 by Ghostkeeper