Ultimaker/Cura #9602: Initial layer speed setting doesn't effect

Mehrad-zh, 2021-04-18:
### Application Version

4.8.0

### Platform

windows 10

### Printer

custom printer

### Reproduction steps

I customized all speed settings in the Speed panel.
I changed the "initial layer speed" and "initial layer print speed" to 15mm/s
The other printing speeds are the same and equal to 30mm/s

### Actual results

Neither in the preview panel in the speed scheme nor in the actual printing process the initial layer doesn't print slower and it performs just as the other layers do.

### Expected results

The initial layer should have been printed slower. actually, it doesn't affect by the "initial layer speed" setting at all

[Desktop.zip](https://github.com/Ultimaker/Cura/files/6330701/Desktop.zip)

fvrmr, 2021-04-20:
Hi @Mehrad-zh thank you for your report.
Does this also happen on other models?

Mehrad-zh, 2021-04-20:
> Hi @Mehrad-zh thank you for your report.
> Does this also happen on other models?

Hi.
Yes. it is not related to the model.

fvrmr, 2021-04-29:
Hi @Mehrad-zh I think this is happening because you are using a custom printer. How did you add this printer to Cura?

Ghostkeeper, 2021-05-05:
It's not about which printer is being used. The printers are just a set of settings. It's just in what the setting values are.

In this case, the Number of Slower Layers setting has been set to 0 by the "ABS 0.8" profile. That setting determines how many layers will be slowed down by Initial Layer Speed. Since it's set to 0, no layers are affected by Initial Layer Speed. As a result, we also hide the setting if this is the case, since Cura 4.9.

closed 2021-05-05 by Ghostkeeper

Mehrad-zh, 2021-05-06:
> It's not about which printer is being used...

That was it. Thank you.