Ultimaker/Cura #9604: Options for max. and min. Adaptive Layer Height?

BloodyRain2k, 2021-04-18:
### Is your feature request related to a problem?

I've trying out the Adaptive Layer Height option a bit and I noticed that it's a bit difficult to make it stay within a certain max. and min. Layer Height.

### Describe the solution you'd like

Profile settings to be added for min. and max. Adaptive Layer Height.

### Describe alternatives you've considered

I managed to solve the max. layer height by editing in a formula in my profile's settings
`adaptive_layer_height_variation = =machine_nozzle_size * 0.6 - layer_height`
which limits the layer height to 0.24 mm with a 0.4 mm nozzle.

What I've not been able to achieve is the min. layer height. When I sliced a model with some fine detail, I saw that the layer height range was always shown with 0.02 mm as thinnest layer in the preview visualization.
By changing `adaptive_layer_height_variation_step` I was able to change where the average ended up, but it still never went above 0.02 mm for the min. scale, so I assume it still ended up using that somewhere.

### Affected users and/or printers

I don't know how widely used the Adaptive Layer Height is, but it could likely benefit people who print either models that have longer continous sections and makers of technical parts.

brunoosti, 2021-04-19:
Yes! This would make much more sense. Sometimes you want to aim at 0.2 layers with larger  .5 nozzle, using its entire lay thikbess range up to 0.4, but don't want to go below 0.1 to save time.

Asterchades, 2021-04-20:
Looks to me like a duplicate of #8174 and #2895. I suspect it will be closed for much the same reasons.

fvrmr, 2021-04-20:
Hi @BloodyRain2k thank you for your feature request. 
This is indeed a duplicate of #8174 and #2895 like @Asterchades mentioned. And I will close this issue indeed also because it is a duplicate and for the same reasons. The reason you why you can read [here](https://github.com/Ultimaker/Cura/issues/8174).
Hope you understand!

closed 2021-04-20 by fvrmr