Ultimaker/Cura #9605: Can´t see a z seam

thestranger0078, 2021-04-18:
### Application Version

4.9.0 beta

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

beta installed to test the new features... can´t see any z seam.... it´s not showed like on your website

### Actual results

tested diffferent parts.... alway the same. no z-seam

### Expected results

show z-seam

thestranger0078, 2021-04-18:
https://user-images.githubusercontent.com/74991234/115151064-982d9100-a06b-11eb-9373-f45e03a270e7.MP4

[image]

fieldOfView, 2021-04-18:
Cura is running in "compatibelity mode", either because you told it to (in the general preferences), or because your GPU only supports that mode. The z seam is only shown in the "full" mode.

GregValiant, 2021-04-18:
Those of us that must work in compatibility mode don't get the fancy stuff.  My old laptop doesn't support OpenGL 4.1 but I have a couple of workarounds.
This image is with the Outer Wall Wipe Distance set to 0.0 and the Shell option in the preview is turned on.  You can barely make out the Z seam and it's on a surface, not hidden in a corner.
[image]

This image shows the Outer Wall Wipe Distance set to 2.0 and the Shell option is turned off.  After viewing where the Z seam is, then I set things back to normal for printing.
[image]

thestranger0078, 2021-04-18:
[image]

thestranger0078, 2021-04-18:
i´m using this on my pc? it´s that not enought for cura??

thestranger0078, 2021-04-18:
> Those of us that must work in compatibility mode don't get the fancy stuff. My old laptop doesn't support OpenGL 4.1 but I have a couple of workarounds.
> This image is with the Outer Wall Wipe [...]

ahhh ok... nice trick..... but first i want to find out why cura 4.9 doesn´t show the seam.... maybe only an setting

fieldOfView, 2021-04-18:
It looks like that GPU should suffice. It could be that the driver is lying to Cura that it cannot create an OpenGL 4.1 context, or you have inadvertently set the option to `Force layer view compatibility mode` in the general preferences.

thestranger0078, 2021-04-18:
> It looks like that GPU should suffice. It could be that the driver is lying to Cura that it cannot create an OpenGL 4.1 context, or you have inadvertently set the option to `Force layer view compatibility mode` in the general preferences.

now i found it in general settings... sorry...... thx a lot. problem solved :)

fvrmr, 2021-04-19:
Since it is solved I will close this issue.

closed 2021-04-19 by fvrmr