Ultimaker/Cura #9606: Hi DPI screen bug, not nice startup screen on ubnuntu, splash screen

avilab84, 2021-04-18:
### Application Version

4.8.0

### Platform

ubunutu

### Printer

none

### Reproduction steps

start cura, see the splash screen

### Actual results

the splash screen is super tiny

### Expected results

it should take more space on screen. Size is in px not in % of screen size. should be fixed.

avilab84, 2021-04-18:
[image]

fvrmr, 2021-04-20:
Hi @avilab84 thank you for your report.
Could you try with the AppImage of Cura 4.9 beta? And see if it happens there?

Ghostkeeper, 2021-05-05:
It is shown in the screenshots that the user was already using the AppImage.

Ghostkeeper, 2021-05-05:
Ah, I misread, the question was to use 4.9.

Nevertheless, I can reproduce this issue on 4.9 as well, by setting my scale to 200% in Ubuntu, so it is a confirmed bug then.

Ghostkeeper, 2021-05-06:
We've talked this over with the developers and consider this to be quite a minor bug. We'll keep it open in this tracker, but it won't get priority for now.

closed 2023-06-10 by github-actions[bot]