Ultimaker/Cura #9607: M107: Fan Off (Deprecated in RepRapFirmware. Use M106 S0 instead)

karl1962, 2021-04-18:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

reprap

### Reproduction steps

Set "Initial Fan Speed" to 0.0% under cooling in CURA
Code that is incorrect for reprap, per there g-code documentation 
M107: Fan Off (Deprecated in RepRapFirmware. Use M106 S0 instead)

> ;LAYER:0
> M107
> 

A workaround is to set the "Initial Fan Speed" to 0.01% this rounds to 0, generating "M106 S0"

[CE3 Linear Pressure advance tower M107.txt](https://github.com/Ultimaker/Cura/files/6331840/CE3.Linear.Pressure.advance.tower.M107.txt)

### Actual results

;LAYER:0
M107

### Expected results

;LAYER:0
M106 S0

fvrmr, 2021-04-20:
Hi @karl1962 thank you for your bug report.
I will discuss this with the team. Keep you posted!

jellespijker, 2021-04-21:
Duplicate of #4605 will close this in favor of that ticket. 
Keep in mind that M107 is deprecated in one firmware version only.

closed 2021-04-21 by jellespijker