Ultimaker/Cura #9608: Combing doesn't work as expected

devydd, 2021-04-18:
### Application Version

Arachne_engine_beta 99.9.1

### Platform

Windows

### Printer

Ender 3

### Reproduction steps

Combing: not in skin (or any other setting)

Box with 3 ribs - there are travels across the model (__box_3 stl file).
[image]

Box with 2 or 1 rib - fine (__box_1 and __box_2 stl files).
[image]

## Cura 4.9.0
fine in both cases no travels across the model.
[image]

STL files: [boxes.zip](https://github.com/Ultimaker/Cura/files/6332025/boxes.zip)
Slicer config file: [creality_ender3_pla_hollow_fast_%234.inst.zip](https://github.com/Ultimaker/Cura/files/6332041/creality_ender3_pla_hollow_fast_.234.inst.zip)

### Actual results

see above

### Expected results

as in Cura 4.9.0 (no travels across the model)

devydd, 2021-04-18:
New finding: with combing mode "Within Infill" setting the "Travel avoid Distance" to a rather large value (5 mm) "fixed" the issue in a rather unexpected way.
[image]

fvrmr, 2021-04-19:
Hi @devydd thank you for your bug report.
Could you share your project files (file - save project) instead of stl's. I want to see your fix and bug myself :)

devydd, 2021-04-19:
Sure. I managed to produce even move curious travel paths:

###  Travel avoid Distance = 0 (travel as above)
3mf: [Travel avoid Distance = 0.zip](https://github.com/Ultimaker/Cura/files/6338816/Travel.avoid.Distance.0.zip)

[image]

---
### Travel avoid Distance = 4.503 (threshold value)

3mf: [Travel avoid Distance = 4.503.zip](https://github.com/Ultimaker/Cura/files/6338795/Travel.avoid.Distance.4.503.zip)
[image]

---
### Travel avoid Distance = 4.504 (above the threshold value)

3mf: [Travel avoid Distance = 4.504.zip](https://github.com/Ultimaker/Cura/files/6338827/Travel.avoid.Distance.4.504.zip)

[image]

---
### Travel avoid Distance = 0 + checked: Outer Before Inner Walls

3mf: [Travel avoid Distance = 0 + OuterBeforeInnerWalls.zip](https://github.com/Ultimaker/Cura/files/6338851/Travel.avoid.Distance.0.%2B.OuterBeforeInnerWalls.zip)

[image]

---
### Travel avoid Distance = 0 + checked: Outer Before Inner Walls + checked: Order Inner Walls By Inset
The strange travels are executing AFTER printing all the lines on a given layer.

3mf: [Travel avoid Distance = 0 +OuterBeforeInnerWalls +OrderInnerWallsByInset.zip](https://github.com/Ultimaker/Cura/files/6338868/Travel.avoid.Distance.0.%2BOuterBeforeInnerWalls.%2BOrderInnerWallsByInset.zip)

[image]

Ghostkeeper, 2021-05-06:
The Travel Avoid Distance >= 4.503mm is expected as that is about the spacing between your walls there. The avoid distance basically says that it's not allowed to come closer than 4.503mm to your model. Since the spacing between the walls is 9mm, anything over 4.5mm should prevent it from going between the walls.

It's also possible that it's different with Order Inner Walls By Inset and Outer Before Inner Walls checked, if that changes the order in which walls are printed.

The combing not working properly with odd numbers of walls was also reported in the discussion thread about the beta, here: https://github.com/Ultimaker/Cura/discussions/9388#discussioncomment-593924 . Developers, refer to CURA-8117.

closed 2024-12-08 by github-actions[bot]