Ultimaker/Cura #9609: Please Add Jerk and Acceleration to the bridging settings

SquirrelEngineering, 2021-04-18:
### Is your feature request related to a problem?

Some types and brands of filament do not bridge as well as others. 

### Describe the solution you'd like

To improve the quality of bridging, please add Jerk and Acceleration settings to bridging. It is something Slic3r has, and I have noticed much better bridging 

### Describe alternatives you've considered

.

### Affected users and/or printers

all

fvrmr, 2021-04-20:
Hi @SquirrelEngineering thank you for your feature request.
I will discuss this with the team. Keep you posted!

jellespijker, 2021-04-21:
@SquirrelEngineering could you explain in more detail why you would think this will have more benefits.

I personally think the benefits are a bit limited, since bridging are per definition long straight lines and jerk applies to corners and turns, while acceleration determines how quick you are at the target speed. Since the bridging lines are long the target speed is most likely achieved anyway.

SquirrelEngineering, 2021-04-21:
Sure. The point where the print is reversing direction is pretty critical in the bridging. I have noticed the beginning of the bridge is often thinner than the rest of the bridge. It is fine all the way until the end where (I assume) it stops extruding and turns around.  In Slic3r, I have been able to fine tune the bridge settings and get excellent bridging.

Ghostkeeper, 2021-05-06:
If you print at a normal 10mm/s skin jerk and 0.4mm line width, it pauses the extrusion for 0.04 seconds even if your acceleration is "infinitely" low. You wouldn't think that is a real problem for extrusion.

Still, if the acceleration is in play that 0.04s can be reduced, but you only have 0.2mm to accelerate so it's _really_ not going to matter a lot for your time difference. Using some napkin mathematics, at a normal acceleration of 3000mm/s² you'd get ~0.011s to accelerate, to a speed of 33mm/s. So your average speed is then (10+33)/2=21.5s, and you'd get that time reduced to 0.019s. Adjusting the acceleration to a way more extreme 6000mm/s² reduces that to 0.016s. So your acceleration adjustment makes a difference of a whole 0.003 seconds. I don't think that makes a difference. The timing between the steps might even remove it altogether.

Have you tried changing the skin pattern to ZigZag? That removes the travel move altogether.

SquirrelEngineering, 2021-05-10:
I think you assume I have a better printer than I do. None of my accelerations are anywhere near 3000. I am printing at around 800-1200.

I have not adjusted skin acceleration. I did not know they were related. I will try that.

Ghostkeeper, 2021-05-18:
The skin acceleration is used for the part of bridging that is skin. The inner/outer wall acceleration is used for the part of bridging that is inner/outer wall.