Ultimaker/Cura #9610: Solid infill ever x layers

MasterCATZ, 2021-04-19:
### Is your feature request related to a problem?

tall towers have weak side support  and in my case allow air to pass through the object where ever leaks are 

my 3d projects are submerged in water and used with pressurized air  90% of the object is infill 

### Describe the solution you'd like

being able to have a solid infill every x layers and have a set amount of layers to do 

ie) every 100 layers do 3x solid layers , this will help add side strength and setup barriers to prevent water / air proceeding through entire object if a leak forms 

### Describe alternatives you've considered

I use solid infill option in prusaslicer but it only does 1 layer so it is not solid enough 

### Affected users and/or printers

unsure but I have seen others asking about this feature on the forums as well so it is not just me 

https://community.ultimaker.com/topic/26962-does-cura-have-a-solid-infill-every-x-layers-function/

GregValiant, 2021-04-19:
You can put in Support Blockers configured for Overlaps.  Make them the size and thickness you want, adjust where each one is in the Z, and one of the settings you can choose is Infill Density.

fvrmr, 2021-04-20:
Hi @MasterCATZ thank you for your feature request.
Can you let us know if the work around of @GregValiant works for you?

MasterCATZ, 2021-04-21:
I have not been able to get the Overlap / Infill density to work and the GUI lags to much attempting it

GregValiant, 2021-04-21:
Here is an example.  You can Ungroup the model and select a mesh modifier to see it's settings.  In this case all the Mesh Modifiers are configured the same with Infill Density at 100%.
[GITHub mesh modifiers.zip](https://github.com/Ultimaker/Cura/files/6350488/GITHub.mesh.modifiers.zip)

MasterCATZ, 2021-04-21:
@GregValiant  thanks , that does do what I am after

I have no idea how to alter your project file to edit meshes everything seems to be selected as 1 unit when I click on it ?
guess I will have to google around for how to ungroup

Using Ubuntu 4.9 beta
I randomly got it to work for me once , I find it very hard for it to start placing the support blocker cubes for editing then unable to edit them latter on as clicking on it removes it 
GUI is very clunky,it seems I have to click the model for support blocker to enable then click off the model then click on the model to place the blocker then edit it before clicking away again  , online tutorials from 2016 show a better more understandable layout 

but I am glad to have finally half gotten mesh edits to work , finally I will be able to put fill exactly where I want them and at the % I want them

GregValiant, 2021-04-22:
Control+left click on the piece you want to select.  Select the button on the left toolbar that has the 4 small icons on it.  (I think the tooltips are broken on that toolbar.)

Ghostkeeper, 2021-05-06:
I thought we'd already gotten feature requests like this before but I can't find it now.

There are workarounds possible, but in the end that goes for almost everything. They aren't really convenient, nor do they allow for this technique to be used by default in a profile. It does allow testing whether it is a feasible technique. I think it could have some value for resistance to shearing, but the problem will be that a single layer might just be sagging too much (especially when using gradual infill) and then not add significant strength.

I'll propose it to the development team to decide what we'll do with planning.

Ghostkeeper, 2021-06-09:
This was discussed with the print engineers of Ultimaker. They raised some concerns that the solid layers cause overextrusion, and different shrinkage strengths such that they cause more harm than good. They've done tests with adding an infill mesh without any walls (just skin) of a single layer thick, but it hasn't been as effective to increase stiffness as the obvious alternative: increase the infill density overall.

As such this is a feature we'll reject. According to our tests, this feature is less effective than the features already implemented in Cura. If it were implemented it should not be used.

We didn't test water permeability, but once water is already among your infill that is really quite a lost cause anyway. If you really need that you'd have to work with the hard-to-use version with infill meshes.

closed 2021-06-09 by Ghostkeeper