Ultimaker/Cura #9611: distance between models for auto-arrange

SamuelSpencer, 2021-04-19:
### Is your feature request related to a problem?

The arrange all models feature places models very close together. I've had a couple of issues with models getting stuck together / being difficult to remove from a raft due to this. Also if one model should fail printing in some way it is more difficult to select that area to stop printing (I use exclude regions in octoprint to do this) due to the difficulty of selecting the correct area to just encompass that one model.

### Describe the solution you'd like

It would be great if the distance between auto arranged models was configurable so that  could choose to have them several millimetres apart. presently I have to manually move them apart after auto-arranging which kind of defeats the purpose.

### Describe alternatives you've considered

the alternative is manually arranging models which is time consuming and may not provide an optimal arrangement

### Affected users and/or printers

everyone who uses arrange all models

fvrmr, 2021-04-20:
Hi @SamuelSpencer thank you for your feature request. 
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-04-26:
We've [increased the spacing](https://github.com/Ultimaker/Cura/commit/83767764db36bd59deb33289fcd3da32d4e2cb69) slightly for the next version, from 1mm to 1.5mm. This brings it closer to what it was in version 4.7 and earlier (which arranged objects on a 2mm resolution pixel grid and made sure the pixels weren't touching).

We don't think that 1mm space (or 1.5mm) is a problem for models getting stuck together. Most printers print at an accuracy of around 0.2mm. For the print to be more than a line width too big would be excessive. And it would have to be over a line width too big on both sides for the prints to start touching even slightly. I don't really know what the raft has to do with this, unless you really need to remove objects one-by-one from the raft rather than pulling the entire raft off in one go. If you do need to pry them off one by one, 1.5mm might make it a bit easier to use a screwdriver to separate them but it's not really going to make a big difference. It is a valid use case then.

Making it configurable seems to be producing bugs in the library we're using to arrange these models. Anything above 1.667mm seems to prevent it from arranging properly even though there is clearly enough space. Perhaps there are then some fixed items on the build plate that overlap or something? I wouldn't know, but it's not as trivial as increasing that number further.

But for these reasons we don't think the benefits outweigh the cost of development. It's a valid feature request, so we'll leave it open, but don't expect it to be developed for the next release.