Ultimaker/Cura #9613: Ramping elephant foot compensation (horizontal expansion initial layer)

nickbuhling, 2021-04-19:
### Is your feature request related to a problem?

The horizontal expansion option for the first layer is great, however often a print needs more than only the first layer to be compensated for an elephant-foot free print. I am aware of proper bed leveling, but in some cases you just need the extra squish and it would be great if the slicer can adjust for it.

### Describe the solution you'd like

Horizontal expansion that ramps over the first x amount of layers. So a bit less (negative) expansion every layer until a certain height. 

### Describe alternatives you've considered

Embedding filets in the model (essentially what this feature would do in the slicer) or adjusting z-height (leading to adhesion problems sometimes). 

### Affected users and/or printers

Everyone who suffers from elephants footing

fvrmr, 2021-04-20:
Hi @nickbuhling thank you for your feature request. 
I don't think this will solve your problem. It would also be complex to look at the difference between the parts that have horizontal expansion and the parts that don't. And setting this for every layer would be difficult.
The best way is like you've mentioned bed levelling.

Let me know what you think!

CCS86, 2021-04-21:
My thought for this would be more simple to execute. Instead of doing horizontal contraction, you could just use a flow modifier for only the outer wall. This would allow full squish for the entire first layer(s), and intentionally under-extruding only the outer wall to help compensate. This could easily be tapered out over n-layers as well.

nickbuhling, 2021-04-23:
Hi @fvrmr, thank you for taking the time to consider my request. I think I either did not explain my suggestion properly, or I am underestimating the technical difficulty ;). Here is a sketch of what I imagine: 

[image]

Now we have the option to set a negative "Initial Layer Horizontal Expansion", which then compensates the first layer. I would suggest adding 1 extra option saying something like "Amount of Initial Layers Horizontal Expansion". This is not optimally formulated I see, but I hope you get the point. Setting it to three would ramp the compensation back to 0 over these 3 layers to create like a bevel. Is it not possible to apply the same method to the other first few layers than with the option that already exists now, but then with reduced values for every layer?

Let me know if this is what you understood already and whether the same answer applies :). 

@CCS86 Thank you for your input too! That might work as well I guess, but I am not sure if it would reduce skin quality maybe?

Ghostkeeper, 2021-06-09:
This was discussed with a print engineer. Both the Cura team and that print engineer think this is a valuable feature to improve dimensional accuracy, especially fit between multiple prints. It's been added to our backlog.

Developers, see issue CURA-8302.

tastyratz, 2022-03-31:
Can I piggyback on this a bit?
I agree here, we need initial layer(s) horizontal expansion for more than a single layer. This can be married to the bottom layers count or an initial layer count and scaled to 0 by the time it gets there, just like fan speed ramp up works today.

Also,
I believe it makes sense to make initial layer horizontal expansion cumulative with horizontal expansion, not independently set.
If horizontal expansion should be set at 0.3mm for part scale and I normally leave 0.2mm for elephants foot, initial layer should be 0.5mm and regular layers 0.3. Otherwise, it creates a surprise brim while initial layer expansion rates tend to be material constant.

gabemorris12, 2022-08-23:
> This was discussed with a print engineer. Both the Cura team and that print engineer think this is a valuable feature to improve dimensional accuracy, especially fit between multiple prints. It's [...]

@Ghostkeeper Is there any way for us to see the backlog? I looked around for a bit but couldn't find anything. The multiple layers for initial horizontal expansion is a feature that I will be constantly checking to see if it has been added.

smartavionics, 2022-08-23:
See https://github.com/smartavionics/Cura/releases for a Cura variant that has this feature since it was recently requested (https://github.com/smartavionics/Cura/issues/181).

gabemorris12, 2023-05-08:
Ok. This feature essentially eliminates elephant's foot, and I am curious as to why it hasn't been implemented in Cura 5 after all this time. I am still using 4.20.3 because of this feature alone, and to my knowledge, no other slicers have this feature. Is there anyway this can be set to a higher priority? This is something that even makersmuse on youtube would recognize. I have been making many parts that are meant to slide fit together, and this feature makes a huge difference.