Ultimaker/Cura #9615: Focus on the object I want print when I zoom in and zoom out

zdfdreamfactory, 2021-04-20:
### Is your feature request related to a problem?

When I have small things to print, switch to 3d view,  I have zoom in to get a clear view, but there is a problem when I zoom in, I can't see my things anymore, I can see my things only on top view, because the things I want print is small

### Describe the solution you'd like

Same like other 3d design software, like CREO, SOLIDWORKS, I want see the 3d object clearly, especially when I add support, I need  a clear preview. 

### Describe alternatives you've considered

just like the solution I say

### Affected users and/or printers

I think it will affect all users, I can see more clearly when I add custom support

fvrmr, 2021-04-20:
Hi @zdfdreamfactory thank you for your feature request. 
You could enable the option `center camera when item is selected.` 
Let me know if that works for you.

zdfdreamfactory, 2021-04-21:
> Hi @zdfdreamfactory thank you for your feature request.
> You could enable the option `center camera when item is selected.`
> Let me know if that works for you.

Thanks, it works, I should have search for the manual, sorry.

closed 2021-04-21 by zdfdreamfactory