Ultimaker/Cura #9617: [4.8.0] Sparse Prime Tower base

Block137, 2021-04-20:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

Slice this project [project.zip](https://github.com/Ultimaker/Cura/files/6344255/project.zip)
The actual cause is still unknown

### Actual results

Base of the Prime Tower is sparse, see attached image:

### Expected results

Prime Tower base completely filled for better adhesion.

fvrmr, 2021-04-22:
Hi @Block137 thank you for your bug report. 
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-05-06:
Investigating what happened to this ticket, it got discussed with some people and it was said that this was thought to be a duplicate but the original needed to be found. However that never happened.

I tried reproducing the issue and found the culprit. It was mixing up a list of paths for the prime tower for each extruder, mismatching the paths with the wrong extruders. I've fixed that up and it should work correctly then for version 4.10 and later.

closed 2021-05-06 by Ghostkeeper