Ultimaker/Cura #9621: Configure setting visibility stopped scrolling to the desired item

eclemens, 2021-04-20:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Creality CR-10s

### Reproduction steps

1. Go to Print Settings (Custom) panel
2. Click on the Section wrench icon or Right click on a setting name then Configure setting visibility 
3. Preferences page in the Setting Visibility window is open

### Actual results

Preferences' Setting Visibility window is open, but the scroll is at top/begging of the list

### Expected results

Preferences' Setting Visibility window is open, and the scroll is down to the setting I selected, as it was working on the previous version of Cura

fvrmr, 2021-04-22:
Hi @eclemens thank you for your bug report. 
I am able to reproduce this and will discuss it with the team. 
Keep you posted!

GregValiant, 2021-04-30:
Hi @fvrmr .  My Setting Visibility dialog has become unusable when "Check All" is ticked.  The window doesn't want to re-populate after scrolling.  Scrolling with the mouse wheel is the same.  Clicking on the up-down arrows on the scroll bar works for 2 or 3 clicks and then it locks up again and takes quite a while to repaint.  The slider causes all checkboxes to disappear for 15 to 20 seconds.

[image]

Ghostkeeper, 2021-05-07:
Hmm, I'm unable to reproduce that problem, @GregValiant :confused: I can click the arrows of the scroll bar all I want without even a visual hiccup. When I drag the scrollbar fast it flickers a bit as it brings the text into render view, but not more than a frame.

Ghostkeeper, 2021-05-07:
This ticket was supposed to be investigated, and if it was an easy fix they'd fix it immediately. But I think they forgot.

I fixed it for the next release, now.

closed 2021-05-07 by Ghostkeeper

fvrmr, 2021-05-07:
@GregValiant please make a new ticket for the setting visibility.

GregValiant, 2021-05-07:
OK.

Ghostkeeper, 2021-05-10:
(New ticket was created here: https://github.com/Ultimaker/Cura/issues/9790)

BloodyRain2k, 2021-05-25:
For anyone who wants this fixed now:
Open the file "%cura_4-9-1_directory%\resources\qml\Cura.qml" in a text editor and scroll to around line 528, it should almost look like this:

`    Connections
    {
        target: Cura.Actions.configureSettingVisibility
        function onTriggered(source)`

What's missing for this part to work is nothing but "source" inside the ( ) of "onTriggered". Change it so that line looks like in the snippet, restart and enjoy.

eclemens, 2021-05-25:
> For anyone who wants this fixed now:
> Open the file "%cura_4-9-1_directory%\resources\qml\Cura.qml" in a text editor and scroll to around line 528, it should almost look like this:
>
> ``` [...]

Thanks, it worked!

commit 912a65b0 (2021-05-07): Add missing source argument to onTriggered handler

Otherwise the source variable is always undefined.

Fixes #9621.