Ultimaker/Cura #9622: Cura 4.9 crashes when opening file.

moo00se, 2021-04-21:
**Application version**
4.9

**Platform**
Windows 10

**Printer**
Any printer

**Reproduction steps**
1. File > Open 
2. Click the open folder icon 
3. 
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Acura crashes

**Expected results**
open up a file. 

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See[cura.log](https://github.com/Ultimaker/Cura/files/6347665/cura.log)s to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**

dragging an object in works. 

In other previous similar issues, I have scene IE and Dropbox being effecting, I don't use those.

Ghostkeeper, 2021-04-21:
Duplicate of https://github.com/Ultimaker/Cura/issues/9554.

fieldOfView, 2021-04-21:
@Ghostkeeper: saving a file != opening a file

We need to see Cura.log to know what is going on.

Ghostkeeper, 2021-04-21:
Fair enough, but I would imagine that both crashes are very related!

fieldOfView, 2021-04-21:
@moo00se, do you have the "Printjob Naming" plugin installed? If so, it could be that you have an older version of that plugin installed. Could you please check if there is an update for the plugin in the Marketplace? The latest version fixes an incompatibility with Cura 4.9, and was published around a month ago.

moo00se, 2021-04-21:
Are you not able to click on the "cura.log" file I posted? 

I do not have that plugin and to add to it I am not logged in when this event occurs.

moo00se, 2021-04-21:
> @Ghostkeeper: saving a file != opening a file
> 
> We need to see Cura.log to know what is going on.

Thank you! I was frustrated to see it that multiple people think these events are related.  I spent too long searching that particular issue and it was no where near relative to my issue.

moo00se, 2021-04-21:
2021-04-21 07:15:40,071 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2021-04-21 07:15:40,092 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-04-21 07:16:37,244 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-04-21 07:16:37,257 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-04-21 07:16:37,267 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-04-21 07:16:37,274 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-04-21 07:16:37,282 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-04-21 07:16:37,291 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-04-21 07:16:37,308 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-04-21 07:16:37,319 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-04-21 07:16:37,331 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-04-21 07:16:37,340 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-04-21 07:16:37,350 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-04-21 07:16:37,360 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-04-21 07:16:37,370 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-04-21 07:16:37,379 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-04-21 07:16:37,389 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-04-21 07:16:37,402 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-04-21 07:16:37,411 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-04-21 07:16:37,420 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-04-21 07:16:37,428 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-04-21 07:16:37,442 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
2021-04-21 07:16:37,449 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-04-21 07:16:37,472 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimulationView
2021-04-21 07:16:37,481 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SliceInfoPlugin
2021-04-21 07:16:37,494 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-04-21 07:16:37,504 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-04-21 07:16:37,538 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-21 07:16:37,540 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-21 07:16:37,542 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-21 07:16:37,544 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-21 07:16:37,545 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-21 07:16:37,547 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-21 07:16:37,549 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-21 07:16:37,550 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-21 07:16:37,556 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin Toolbox
2021-04-21 07:16:37,566 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CameraTool
2021-04-21 07:16:37,577 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MirrorTool
2021-04-21 07:16:37,591 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RotateTool
2021-04-21 07:16:37,605 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ScaleTool
2021-04-21 07:16:37,613 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SelectionTool
2021-04-21 07:16:37,626 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin TranslateTool
2021-04-21 07:16:37,634 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin TrimeshReader
2021-04-21 07:16:37,651 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin UFPReader
2021-04-21 07:16:37,660 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin UFPWriter
2021-04-21 07:16:37,670 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin UltimakerMachineActions
2021-04-21 07:16:37,782 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin UM3NetworkPrinting
2021-04-21 07:16:37,795 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin UpdateChecker
2021-04-21 07:16:37,795 - INFO - [JobQueWorker [0]] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2021-04-21 07:16:37,831 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin USBPrinting
2021-04-21 07:16:37,848 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin VersionUpgrade21to22
2021-04-21 07:16:37,858 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin VersionUpgrade22to24
2021-04-21 07:16:37,867 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin VersionUpgrade25to26
2021-04-21 07:16:37,875 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin VersionUpgrade26to27
[... cut ...]
2021-04-21 07:16:59,030 - WARNING - [MainThread] UM.Logger.warning [134]: Starting a new sync while previous sync was not completed
{'CloudOutputDeviceManager': 0, 'CloudPackageChecker': 0}
2021-04-21 07:16:59,699 - DEBUG - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._handleCompatibilityData [127]: Discrepancy found between Cloud subscribed packages and Cura installed packages
2021-04-21 07:17:08,972 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

Sekisback, 2021-04-21:
[cura.log](https://github.com/Ultimaker/Cura/files/6352004/cura.log) may this one can help to, same on my side

fieldOfView, 2021-04-21:
That last cura.log shows the problem I mentioned above, which is caused by an outdated version of the "Printjob Naming" plugin. Please check the Marketplace "Installed" tab for an update.

fieldOfView, 2021-04-21:
It looks like @moo00se and @Sekisback suffer from different issues.

The log @moo00se posted does not contain an obvious crash. This means that the crash happens outside the code that is part of this project, making this a harder to diagnose issue (and probably a duplicate of #9554 after all).

moo00se, 2021-04-21:
> It looks like @moo00se and @Sekisback suffer from different issues.
> 
> The log @moo00se posted does not contain an obvious crash. This means that the crash happens outside the code that is part of this project, making this a harder to diagnose issue (and probably a duplicate of #9554 after all).

then I'm pretty clueless hahah, thanks for clarifying it.

moo00se, 2021-04-22:
Cura, was also making zBrush crash when using the "File Explorer" to import files. Uninstalled Cura, working just fine. I'll probably keep it that way. For someone who tries to be as basic as possible, I find it disappointing that this 4.9 was rolled out when it is clearly not ready. Faith lost in Cura producers.

HeavyMetalGuy, 2021-04-23:
Had the same problem, opened Cura as admin, problem solved (at least for me).

Sekisback, 2021-04-23:
> 
> 
> [cura.log](https://github.com/Ultimaker/Cura/files/6352004/cura.log) may this one can help to, same on my side

OK I checked the Plugins and after I disabled  "CustomPrintjob Naming" everything works fine in case. So it wasnt a Cura issue

fieldOfView, 2021-04-23:
@Sekisback If you want the plugin back, make sure to update to its latest version which fixes the problem you had.

Sekisback, 2021-04-23:
> @Sekisback If you want the plugin back, make sure to update to its latest version which fixes the problem you had.
yes thank you, i had cura4.8 with all needed plugins and after i got it working i had no issue to change the running system.
I switched to 4.9 and updated all the plugins, so from my point of view it runs stable now, like before.

JoeEnderman, 2021-04-25:
I have the same issue, I may switch to 4.8 until the issue is resolved.

MariMakes, 2023-05-24:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, 

I stumbled upon your issue regarding cura crashing when opening a file. 
It reminds me of an issue with Chitubox [that required this work-around. ](https://support.makerbot.com/s/article/1667412729970)

@moo00se, can you confirm if this issue is still present for you in the latest 5.3.1?
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-06-08 by github-actions[bot]