Ultimaker/Cura #9624: com port problem

JAPMedia, 2021-04-21:
This program has serious problems with the COM ports which are opened on every startup and remain open. 
I use 6 com ports to control my radios and HAM Radio amplifiers and it is each time the casino when I start this program. 
Why is the configuration of the COM port not defined in the configuration (as is the rule of common sense) instead of opening them all?
Hope this will be changed quickly. In order to be able to use other devices at the same time.

DivingDuck, 2021-04-21:
+1
An issue since years and often communicated but constantly ignored.

nallath, 2021-04-21:
> +1
> An issue since years and often communicated but constantly ignored.

Not ignored, just not fixed. There is a difference. 

As mentioned before, this doesn't have priority for us to fix. It's not likely that we will spend time on this in the near future. But since this is an open source project, we would more than welcome pull requests that fix this issue. 

As it stands, this is a duplicate of https://github.com/Ultimaker/Cura/issues/3449

closed 2021-04-21 by nallath

JAPMedia, 2021-04-21:
Ok Nallath, at less your responded thanks for this. Hope you can fix this in near future because all my devices get crazy when I launch Cura, In the mean time I use another opensource slicer who don't have this bug.
Good luck

fieldOfView, 2021-04-21:
You can disable the "USB Printing" plugin in the Installed tab of the Marketplace. This will stop Cura from touching your serial ports entirely.

JAPMedia, 2021-04-21:
> 
> 
> You can disable the "USB Printing" plugin in the Installed tab of the Marketplace. This will stop Cura from touching your serial ports entirely.

Thanks man, you make my day. Also now it load 50% faster.

rdarden, 2024-12-06:
Thanks for sharing the workaround. 

I can confirm this is still a problem with Ultimaker Cura app version 5.9.0 on MacOS with a Yaesu FTDX-10 radio connected via USB, and the workaround is still effective.