Ultimaker/Cura #9625: [4.9 beta to 4.9.0] Beta says no new version available

printingotb, 2021-04-21:
**Application version**
4.9 beta to 4.9.0

**Platform**
win7

**Printer**
custom fff

**Reproduction steps**
Start cura 4.9 beta = no message to update
Check for updates = no new version available

**Screenshot(s)**
[image]

jellespijker, 2021-04-21:
This is unfortunately expected behavior since the version is already defined as 4.9 in the beta version. We are considering broadcasting the new beta releases in the future. Imo this is something that would need to be addressed then as well.

fvrmr, 2021-04-22:
We have a ticket on our backlog that is related to the updater. I will add this to the ticket.
Developers see CURA-7689

nallath, 2021-07-26:
https://github.com/Ultimaker/Cura/pull/10189 && https://github.com/Ultimaker/Uranium/pull/715

closed 2023-05-24 by MariMakes