Ultimaker/Cura #9628: Cura 4.9.0 very SLOW

SilentWraith5, 2021-04-21:
**Application version**
Version 4.9.0

**Platform**
Windows 10 Laptop
i7 10510u 
16GB RAM
UHD graphics 

**Printer**
Ender 3

**Reproduction steps**
1. Opening model
2. Clicking on the model is really slow 

**Actual results**
When you click on the model it takes a very long time for it to load before you can move it. The whole software is slow compared to 4.8.0

**Expected results**
When clicking on the model it takes forever for it to register and then when trying to multiply the model the small window doesn't load for 15 seconds  

**Additional information**
The whole software is running poorly since upgrading to 4.9.0 and everything seems to be running very slowly and unusable

jellespijker, 2021-04-21:
Can you provide us with the log files. See the help menu -> configuration folder

SilentWraith5, 2021-04-21:
[cura.log](https://github.com/Ultimaker/Cura/files/6350065/cura.log)

nallath, 2021-04-21:
Is it still slow if you disable the sidebar GUI plugin? I'm seeing a lot of warnings in the logs that come from that.

fieldOfView, 2021-04-21:
'''
2021-04-21 17:13:39,821 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/resources/themes/cura-light/styles.qml:335: TypeError: Cannot read property 'getSize' of null
2021-04-21 17:13:39,824 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/resources/themes/cura-light/styles.qml:338: TypeError: Cannot read property 'getColor' of null
'''
I don't see how these would be caused by SidebarGUI. And there's about 450 consecutive errors just like that, taking about 3 seconds.

nallath, 2021-04-22:
There were also a bunch of warnings about the onFoo attributes from the sidebar plugin. 

I can't explain why the theme is not working in this case.

SilentWraith5, 2021-04-23:
I’ve found the issue I was having. The sidebar GUI was the cause for making it run very slowly and it’s fixed. But why would it be doing that?

fieldOfView, 2021-04-24:
> The sidebar GUI was the cause for making it run very slowly and it’s fixed.

What do you mean by "and it's fixed"? Did installing a newer version of the plugin fix it, or did you disable/uninstall the plugin to fix it?

> But why would it be doing that?

I wish I knew, then I could improve the sidebar gui plugin to not be doing that. I don't think it's the warnings @nallath is referring to.

Anyway, if you could open an issue at https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues about the slowness caused by the plugin, that would be great.

SilentWraith5, 2021-04-24:
I uninstalled it and it seems to be running much better now

nallath, 2021-04-25:
There are two sets of errors that I see with the sidebar GUI in the logs:

```
2021-04-21 18:17:22,061 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users/lzpro/AppData/Roaming/cura/4.9/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-21 18:17:22,076 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users/lzpro/AppData/Roaming/cura/4.9/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-21 18:17:22,076 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users/lzpro/AppData/Roaming/cura/4.9/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-21 18:17:22,076 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users/lzpro/AppData/Roaming/cura/4.9/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
```

And 
```
SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:24:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
```

The first one could be causing quite a bit of slowdown. I don't think the second one will cause a lot of issues.

fieldOfView, 2021-04-30:
I have made a PR that I think fixes the hundreds of qml warnings per second in styles.qml. They may be much more prevalent with SidebarGUI, but I also see them without SidebarGUI.

See https://github.com/Ultimaker/Cura/pull/9732. If there's going to be a Cura 4.9.1, I think this fix should be part of that.

Ghostkeeper, 2021-05-11:
That PR is merged and released with Cura 4.9.1. The majority of the spam is removed by fieldOfView's contribution. Since the remainder seems to be in the SidebarGUI plug-in then, we'll close this issue.

We do have another issue to clean up more of these QML warnings, removing technical debt. It's quite high up in the priority. That should improve it more, but this source of slowness is better with 4.9.1.

closed 2021-05-11 by Ghostkeeper

fieldOfView, 2021-06-23:
For the benefit of other people stumbling upon this thread, disabling the FancyZones PowerToy seems to fix this problem.