Ultimaker/Cura #9629: Cura 4.9 keeps crasching

FrankBreda, 2021-04-21:
I using cura 4.9 now and every time i put in a new file to slice cura crashes.

windows 10
ender 3 v2

SilentWraith5, 2021-04-21:
I'm having the same issue

fieldOfView, 2021-04-21:
Without a logfile, there's no way to diagnose this.

bizzlerulez, 2021-04-21:
I have the same issue 
Error Traceback: `Traceback (most recent call last):
  File "C:\Users\Derya\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\Derya\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'
`
Logs:
 ` Thread 0x00004ad0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005358 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002cc8 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003f08 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000026bc (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000f18 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003810 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000030ac (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000268c (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002dc0 (most recent call first):
  File "threading.py", line 302 in wait
[... 128 lines cut ...]
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
 `

fieldOfView, 2021-04-21:
@bizzlerulez, Your crash is caused by the "Printjob Naming" plugin. It seems you have an older version of that plugin installed. Could you please check if there is an update for the plugin in the Marketplace? The latest version fixes this particular issue, and was published around a month ago.

@FrankBreda, unfortunately github strips attachments from emails, so I can't check if your crash is caused by the same plugin. Just to be sure, could you also check the "Installed" tab of the Marketplace for an updated version?

FrankBreda, 2021-04-21:
2021-04-17 18:40:14,018 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-04-17 18:40:14,033 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2021-04-17 18:40:14,033 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.9.0-BETA
2021-04-17 18:40:14,033 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2021-04-17 18:40:14,033 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2021-04-17 18:40:14,049 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-04-17 18:40:14,049 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-04-17 18:40:14,065 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-04-17 18:40:14,080 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-04-17 18:40:14,082 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-04-17 18:40:14,082 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2021-04-17 18:40:14,082 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2021-04-17 18:40:14,082 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49675
2021-04-17 18:40:14,082 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-04-17 18:40:14,082 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-04-17 18:40:14,098 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-04-17 18:40:14,098 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-04-17 18:40:14,098 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2021-04-17 18:40:14,098 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-04-17 18:40:14,098 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-04-17 18:40:14,113 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-04-17 18:40:14,129 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-04-17 18:40:14,129 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-04-17 18:40:14,145 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-04-17 18:40:14,207 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-04-17 18:40:14,207 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-04-17 18:40:14,223 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-04-17 18:40:14,238 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-04-17 18:40:14,254 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-04-17 18:40:14,270 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-04-17 18:40:14,285 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-04-17 18:40:14,285 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-04-17 18:40:14,301 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-04-17 18:40:14,317 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-04-17 18:40:14,332 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-04-17 18:40:14,332 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-04-17 18:40:14,348 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-04-17 18:40:14,363 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
2021-04-17 18:40:14,379 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-04-17 18:40:14,426 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimulationView
2021-04-17 18:40:14,442 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SliceInfoPlugin
2021-04-17 18:40:14,457 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-04-17 18:40:14,473 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-04-17 18:40:14,520 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-17 18:40:14,520 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-17 18:40:14,551 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-17 18:40:14,551 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
[... cut ...] ="True" -s wipe_hop_enable="False" -s machine_nozzle_heat_up_speed="2.0" -s clean_between_layers="False" -s wipe_retraction_retract_speed="25" -s alternate_extra_perimeter="False" -s cool_fan_speed="100.0" -s machine_steps_per_mm_y="50" -s speed_equalize_flow_enabled="False" -s support_infill_rate="20" -s speed_z_hop="5" -s raft_surface_jerk="8" -s skin_edge_support_thickness="0" -s brim_line_count="20" -s material_end_of_filament_purge_speed="0.5" -s magic_fuzzy_skin_point_density="1.25" -s sup

fvrmr, 2021-04-22:
Hi @FrankBreda could you please add your log file as a document in a comment here on Github. Sending it as a e-mail doesn't work, because it seems like there is a lot missing.

FrankBreda, 2021-04-22:
I think i found the problem with chitubox

nallath, 2021-04-22:
> I think i found the problem with chitubox

Cool! Whould you mind sharing what you found? It could be that there is something that we could / should change so that it's fixed for everyone.

FrankBreda, 2021-04-22:
Ive deleted chitubox from my pc and now it works

nallath, 2021-04-22:
> Ive deleted chitubox from my pc and now it works

What made you try that? Or were you just trying to remove random other SW to see what fixed this issue?

FrankBreda, 2021-04-22:
I saw something on the messageboard about it so thats what i tried.

BarryWillick, 2021-04-28:
I had the same issue that caused a crash every time I attempted to load a file.  Updated all of my plugins and the issue went away.  This was very frustrating and time-consuming to discover the solution.  There should be a more elegant way of handling an outdated plugin than to crash.

fieldOfView, 2021-04-28:
The plugin that caused the crash (one of mone, sorry) was updated to prevent the crash more than a month before the release of Cura 4.9. 

Updating plugins should be more straightforward.

Ghostkeeper, 2021-05-07:
I'll close this because the root issue has already been determined and fixed in the plug-ins that was causing them.

If there's ideas on how to make it easier to update plug-ins, I'm all ears. We could implement a button to batch-update all of them, maybe, but if people don't pay attention to the number in the top-right corner of the screen, I don't think they're likely to push that button either.

closed 2021-05-07 by Ghostkeeper

fieldOfView, 2021-05-07:
> If there's ideas on how to make it easier to update plug-ins, I'm all ears.

The Marketplace could be fixed so it does not take around a minute to open and populate.
The UpdateChecker message that pops up to notify the user of a new version of Cura could be enhanced to also pop up for plugin updates.
There could be an "available updates" tab in the Marketplace.

fieldOfView, 2021-05-07:
You might even want to have an option to automaticly update plugins, especially for "Essentials" approved plugins.

Ghostkeeper, 2021-05-10:
> The Marketplace could be fixed so it does not take around a minute to open and populate.

This is indeed on our backlog and I agree that would certainly help! It's very annoying.

> The UpdateChecker message that pops up to notify the user of a new version of Cura could be enhanced to also pop up for plugin updates.

This is the type of change I wouldn't like to see personally. It's a bit of a subjective issue, but usually it's not a problem to have outdated plug-ins and then getting regular pop-ups about it is kind of annoying.

> There could be an "available updates" tab in the Marketplace.

That's a nice idea. I think I've heard it before (was it you who suggested that?). I can put that on our CCB.

Maybe this discussion should be moved to https://github.com/Ultimaker/Cura/issues/6658 though.

konskarm, 2021-05-28:
@fieldOfView we now have a ticket in our backlog for adding a preference to automatically update plugins. Devs see CURA-8274.