Ultimaker/Cura #9630: New feature show "starts" in Linetype not available

spanzetta, 2021-04-21:
I don't see this option.. but I am in "compatibility mode" because of "old" graphics card..
Is it that normal?
Thanks

[image]

jellespijker, 2021-04-21:
Then you're out of luck unfortunately.
Due to the way in which the rendering is handled.
This isn't an easy fix unfortunately and it is unlikely that we will ever get the time to add this to the compatibility mode.

closed 2021-04-21 by nallath

spanzetta, 2021-05-16:
Update: after a Windows Update (which eventually fixed ATI/AMD graphics driver) and with Cura 4.9.1 I now can see "starts"
Don't which of the 2 fixed the problem.. but I am anyway happy to enjoy this new very nice feature! even with a very old graphics card (HP 4530s with Intel HD3000 + AMD HD7400M)
Thanks