Ultimaker/Cura #9631: [4.9.0] Preview crash on MacOS 10.13.6

jwrw, 2021-04-21:
**Application version**
4.9.0

**Platform**
MacOS High Sierra (10.13.6)
iMac (27-inch Mid 2010)
Processor 2.93 GHz Intel Core i7
Memory 12GB
Graphics ATI Radeon HD 5750

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Load simple model
2. Slice
3. Go to preview

**Screenshot(s)**
No screen - application froze then crashed

**Actual results**
Application crashed when trying to display the sliced model

**Expected results**
Sliced model preview

**Project file**
This has happened with all models I have tried

**Log file**
Attached

**Additional information**
Tried both an upgrade from 4.8 by leaving originals in place and dragging new app to /Applications - preview crashed
Removed 4.9
Removed all added plugins, deleted 4.8 app and removed cura from ~/Library/Application Support/ and re-copied 4.9 to /Applications - again crash on preview
[cura.log](https://github.com/Ultimaker/Cura/files/6350274/cura.log)

ravgu, 2021-04-21:
I Confirm that preview is crashing on Mac OS 10.13.6 on a iMac (21.5-inch, Mid 2011) when selecting preview. However sliced a file and printed it successfully. but each time i try to preview the program crashes.

colatuttorf, 2021-04-22:
I have the same 27" iMac detailed above running the same MacOS and Cura and can report the same issue.  100% crash rate when clicking Preview, which makes the software unusable and crashes the system.

Reverted to 4.8.0 and appears functional again.

jwrw, 2021-04-22:
It's a retrograde step but if you go into Preferences>General>Viewport behaviour then you can select 'Force layer view compatibility mode', which renders each layer as a 2D plane instead of extruded 'sausages'.  It's not nice, but at least it doesn't seem to crash.  You also don't seem to get the playback option (if you ever really used that).

orkide11, 2021-04-22:
Idem pour moi.
Installation 4.8 ça marche !

otitserip, 2021-04-22:
same crash on the same system... back to 4.8.

jwrw, 2021-04-22:
Checking console it appears the GPU is crashed and has to reset.

fvrmr, 2021-04-23:
Hi @jwrw thank you for your bug report. 
Also I am wondering where did you get this info from?

> qtdiag
> Qt 5.9.7 (x86_64-little_endian-lp64 shared (dynamic) release build; by Clang 4.0.1 (tags/RELEASE_401/final)) on "cocoa"
> OS: macOS High Sierra (10.13) [darwin version 17.7.0]

I will discuss this with the team. Keep you posted. 

Also good to know that there is a work around.

jwrw, 2021-04-23:
Hi - after digging around a little more I can see the app uses its own QT library (5.10?).  The qtdiag output is from the terminal prompt on my machine so is probably not relevant.  Sorry if this has caused any confusion.

colatuttorf, 2021-04-23:
The release notes for 4.9.0 say QT has been upgraded to 5.15 (6.0 is the latest version) so 5.9.7/5.10 are quite a few releases back. 

Is there any way to verify which build of QT cura is actively using?  There may be some dependencies which are holding the QT version back on our High Sierra Macs, that we could fix to get cura at least using the latest QT it supports.

jwrw, 2021-04-24:
I tried replacing Contents/Resources/plugins/SimulationView/layers3d.shader with the one from 3.8 and it no longer crashes (it may be throwing other errors internally - but seems to work OK)
I can see most the latest changes are to support the start marker function so I'll see if there is anything obviously wrong (not played with shaders before tho').  
[and checking the QT stuff - Cura is using all its own QT libraries and all the ones I looked at are 5.15 - so no problem there]

jwrw, 2021-04-24:
I believe I have a fix #9682

fvrmr, 2021-04-26:
We've created a ticket for this issue and linked your fix @jwrw 
Developers see CURA-8189

evtrados, 2021-04-28:
> The release notes for 4.9.0 say QT has been upgraded to 5.15 (6.0 is the latest version) so 5.9.7/5.10 are quite a few releases back.
>
> Is there any way to verify which build of QT cura is actively [...]

As of version 4.9.0 Cura uses Qt 5.15.2 (before it was 5.10), which according to Qt's compatibility notes supports macOS versions from 10.13 and higher. Thus, I would not expect any crash on your system, unless there are issues that Qt has with other interrelated packages we use in Cura.

Ghostkeeper, 2021-05-07:
The fix is merged for version 4.9.1. Thanks! We'd probably never have found this ourselves.

closed 2021-05-07 by Ghostkeeper