Ultimaker/Cura #9633: [Arachne Beta] Multiply objects does not work

rreilink, 2021-04-21:
**Application version**
Arachne Beta

**Platform**
Mac OS 10.15.4

**Printer**
UM3

**Reproduction steps**
1. Place an object
2. Right-click, select 'multiply selected objects'
3. click ok

**Actual results**
'Placing Objects' , ' Multiplying and placing objects' appears, but no new objects appear, and the 'Placing Objects' popup keeps present

**Expected results**
New objects should appear

**Log file**
2021-04-21 13:08:23,304 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Failed to arrange the models
2021-04-21 13:08:23,307 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-04-21 13:08:23,309 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1920, in _readMeshFinished
2021-04-21 13:08:23,312 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 125, in arrange
2021-04-21 13:08:23,315 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 76, in findNodePlacement
2021-04-21 13:08:23,317 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 176, in intersectionConvexHulls
2021-04-21 13:08:23,320 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 676, in intersection
2021-04-21 13:08:23,322 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 64, in __call__
2021-04-21 13:08:23,324 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 18, in _validate
2021-04-21 13:08:23,327 - ERROR - [MainThread] UM.Logger.logException [116]: ValueError: Null geometry supports no operations
2021-04-21 13:08:23,713 - DEBUG - [Thread-2] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for custom took 0.32300400733947754 seconds.
2021-04-21 13:08:23,831 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura Arachne Beta.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2021-04-21 13:08:24,128 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///Applications/Ultimaker Cura Arachne Beta.app/Contents/MacOS/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2021-04-21 13:08:24,259 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 8.2s
2021-04-21 13:08:27,194 - DEBUG - [Thread-2] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for ultimaker3_extended took 1.9609310626983643 seconds.
2021-04-21 13:08:28,077 - DEBUG - [Thread-2] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for felix3 took 0.37451910972595215 seconds.
2021-04-21 13:08:28,080 - DEBUG - [Thread-2] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2021-04-21 13:08:32,935 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-21 13:11:02,404 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura Arachne Beta.app/Contents/MacOS/resources/shaders/toolhandle.shader]...
2021-04-21 13:11:06,351 - ERROR - [Thread-4] UM.Logger.logException [112]: Exception: Job <cura.MultiplyObjectsJob.MultiplyObjectsJob object at 0x60000e9b4a18> caused an exception
2021-04-21 13:11:06,356 - ERROR - [Thread-4] UM.Logger.logException [116]: Traceback (most recent call last):
2021-04-21 13:11:06,361 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 132, in run
2021-04-21 13:11:06,364 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/MultiplyObjectsJob.py", line 71, in run
2021-04-21 13:11:06,367 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 125, in arrange
2021-04-21 13:11:06,370 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 76, in findNodePlacement
2021-04-21 13:11:06,372 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 176, in intersectionConvexHulls
2021-04-21 13:11:06,375 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 676, in intersection
2021-04-21 13:11:06,379 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 64, in __call__
2021-04-21 13:11:06,382 - ERROR - [Thread-4] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 18, in _validate
2021-04-21 13:11:06,384 - ERROR - [Thread-4] UM.Logger.logException [116]: ValueError: Null geometry supports no operations
2021-04-21 13:11:06,573 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2021-04-21 13:11:06,788 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-04-21 13:11:06,801 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-04-21 13:11:07,220 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.7s
2021-04-21 13:11:16,866 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-21 13:17:56,289 - ERROR - [Thread-1] UM.Logger.logException [112]: Exception: Job <cura.MultiplyObjectsJob.MultiplyObjectsJob object at 0x60000e8f7e98> caused an exception
2021-04-21 13:17:56,292 - ERROR - [Thread-1] UM.Logger.logException [116]: Traceback (most recent call last):
2021-04-21 13:17:56,295 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 132, in run
2021-04-21 13:17:56,298 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/MultiplyObjectsJob.py", line 71, in run
2021-04-21 13:17:56,302 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 125, in arrange
2021-04-21 13:17:56,305 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/cura/Arranging/Nest2DArrange.py", line 76, in findNodePlacement
2021-04-21 13:17:56,309 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/arachne/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 176, in intersectionConvexHulls
2021-04-21 13:17:56,315 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 676, in intersection
2021-04-21 13:17:56,324 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 64, in __call__
2021-04-21 13:17:56,333 - ERROR - [Thread-1] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/arachne/inst/lib/python3.5/site-packages/shapely/topology.py", line 18, in _validate
[... cut ...]
2021-04-21 13:17:56,759 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-04-21 13:17:56,770 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-04-21 13:17:57,195 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.7s
2021-04-21 13:18:05,877 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

nallath, 2021-04-21:
Could you also share the model that you have the problem with? I've not been able to reproduce it. Based on the logs it looks likes something that is caused by a specific model.

rreilink, 2021-04-21:
Here's the model:

[Body1.stl.zip](https://github.com/Ultimaker/Cura/files/6351009/Body1.stl.zip)

fvrmr, 2021-04-22:
Hi @rreilink thank you for your model.
I am not able to reproduce it with your model. Does it happen on every model?

rreilink, 2021-04-22:
Yes, it does with every model I have tried so far

Vandresc, 2021-04-23:
This is a bug occurring only on os x, we did fix it in 4.9 but unfortunately still in arachne beta. CURA-7986

Ghostkeeper, 2021-05-07:
Indeed, this will be fixed in the next Arachne release then as well.

closed 2021-05-07 by Ghostkeeper