Ultimaker/Cura #9634: Slider progress bar don't move

MattXcz, 2021-04-21:
**Application version**
4.9.0
**Platform**
Windows 10 x64

**Printer**
Ender 3 Proa

**Reproduction steps**
1. Import STL file
2. Click slice

**Actual results**
Slider progress bar wont move, cant slice

**Expected results**
(What should happen after the above steps have been followed.)
Slice model

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

Premision to process denied

**Additional information**
(Extra information relevant to the issue.)

jellespijker, 2021-04-21:
Can you share your log file as well?

MattXcz, 2021-04-21:
I am sorry. I installed version 4.8, which works. I deleted old folder with logs

jellespijker, 2021-04-21:
That is not a lot of information to go on.
Do you have a firewall enabled (NordVPN maybe)?

MattXcz, 2021-04-21:
I am on secured wlan with some port restriction, etc. Does slicer process need to access some external service ?

flashbackpa, 2021-04-22:
Hi, just to inform @MattXcz that I had the same problem, and the root cause was:

2021-04-22 12:18:40,266 - ERROR - [MainThread] UM.Backend.Backend._runEngineProcess [147]: Couldn't start back-end: Operating system is blocking it (antivirus?): [WinError 87]

As soon as I removed cura-engine task from antivirus (in my case Comodo), the slicing process started.

Bye

MattXcz, 2021-04-23:
It looks like Fidelis Endpoint is blocking this process, but since I am slicing on corporate laptot, I can't do anything about it. 
Cura 4.8.0 works.

Ghostkeeper, 2021-05-07:
> I am on secured wlan with some port restriction, etc. Does slicer process need to access some external service ?

Cura opens a COM port to 127.0.0.1:49674 to communicate between Cura's front-end (which loads the scene and settings) and the slicing back-end (which transforms those to g-code). Since it is a localhost address, it doesn't need communication with some external service. However, some firewalls and VPN clients still block it.

This is a way to make the front-end completely immune to crashes from the slicing back-end. If it were a Python subprocess, a segfault from CuraEngine could crash the application. And it doesn't require 400MB models to be transferred over stdin either.

If Cura 4.8 works but 4.9 doesn't, then likely whatever is blocking 4.9 didn't update its threat model yet. The way Cura starts CuraEngine and communicates with it hasn't changed in the past 5 years.

Since an external force is actively blocking part of our application, there is nothing we can realistically do to help you with this case. Sorry! I hope that antivirus or VPN client updates its definitions soon then.

closed 2021-05-07 by Ghostkeeper