Ultimaker/Cura #9638: Slow/intermittent slicing for big/complex objects

guate6, 2021-04-21:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

When slicing large complex models, it begins to slice (CPU at 100%), but then lags for a while. Then at some random time later it picks back up (CPU at 100%), and finishes. This process has taken up to "overnight" to come back to a sliced model. 
I mostly use the 0.4mm core to print. The problems arise when I'm tweaking support blockers and have to check their placement and results after each tweak. I slice each time and it can take an hour for one model. It always happens the same way: The status bar gets to about 50% on the "slicing" and CPU usage goes down to about 9-14%, stays here for a while, then it'll pick back up and finish at 100% CPU. 

The only workaround I've been able to get to work is working with the 0.8mm core, placing support blockers, slicing, tweaking, slicing, etc. Then when finally ready, doing it all with the 0.4mm core and simply waiting a long time for this to work.

Here's one model that takes a bit time (under an hour) to slice (this was a free file from thingiverse, so I don't mind sharing). I have other more complex files, but cannot share them here as I purchased them. I can share them direct with a developer if needed.

https://www.dropbox.com/s/45q5huanqin83mc/z%3D0.1%20%2816d%2011h%2022m%20%2437.22%29%20PLA.3mf?dl=0

Using an Intel i9-9900KS, RTX 3090, and other up-to-date components
[image]
[image]

### Actual results

Slicing takes way too long. Processor ramps up, slows down for the majority of the time, and ramps up again a while later to finish. 

### Expected results

Max CPU usage or GPU usage for the entire slice, making full use of all cores and threads to slice an item/object as quickly as possible.

guate6, 2021-04-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jellespijker, 2021-04-21:
Could you also share your log file?

guate6, 2021-04-21:
How do I do that?

Also I'm happy to share some of the other paid files with someone directly,
but not publicly. Please advise.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

jellespijker, 2021-04-21:
You can send them to me j.spijker@ultimaker.com 

You can find the log files in your configuration folder.
[image]

guate6, 2021-04-21:
I did that a couple times and nothing is showing up.
Here are more complex files I've worked on that take time to slice:

https://www.dropbox.com/s/fk4dwtmjbudxv49/All%20%28minus%20demon%29%20z%3D0.1%20%2816d%2017h%2016m%20%2465.15%29%20PLA.3mf?dl=0

https://www.dropbox.com/s/umbl913bx7906n5/Demon%20z%3D0.1%20%2811d%203h%206m%20%2419.18%29%20PLA.3mf?dl=0

And also note it's not about these particular files themselves, but rather
the behavior of Cura with larger more complex files where it'll
start/pause/resume/finish.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

jellespijker, 2021-04-21:
Thanks for the share. You mentioned that these are paid files. Keep in mind that you shared them publicly now.

jellespijker, 2021-04-21:
Can you try the steps describe here https://github.com/Ultimaker/Cura#logging-issues to obtain the log files

guate6, 2021-04-21:
Thank you for letting me know. Can you please verify that I've deleted
those files properly?

jellespijker, 2021-04-22:
Yes the files are deleted on GitHub.
Do you need additional help with obtaining the log files?

guate6, 2021-04-22:
Got the folder open, but not sure what all you're looking for.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

jellespijker, 2021-04-23:
There should be a file called `cura.log` This is a plain text file with information that is useful to us. Such as which plugins are loaded, how much time certain operations took. Which problems and exceptions Cura encountered. This will help us identify possible bottlenecks.

```
2021-04-23 17:00:37,456 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin PrepareStage
2021-04-23 17:00:37,459 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin MonitorStage
2021-04-23 17:00:37,465 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin XRayView
2021-04-23 17:00:37,468 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin 3MFWriter
2021-04-23 17:00:37,470 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin SliceInfoPlugin
2021-04-23 17:00:37,473 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin PreviewStage
2021-04-23 17:00:37,479 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin SimulationView
2021-04-23 17:00:37,482 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin MachineSettingsAction
2021-04-23 17:00:37,487 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin XmlMaterialProfile
2021-04-23 17:00:37,489 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin SolidView
2021-04-23 17:00:37,491 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin CuraProfileReader
2021-04-23 17:00:37,494 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin UFPWriter
2021-04-23 17:00:37,497 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin GCodeWriter
2021-04-23 17:00:37,501 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin ImageReader
2021-04-23 17:00:37,507 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [470]: Loaded plugin CuraDrive
2021-04-23 17:00:37,507 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [395]: Loading all plugins took 0.8479971885681152 seconds
2021-04-23 17:00:37,512 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2021-04-23 17:00:37,521 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.008732795715332031
2021-04-23 17:00:37,523 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [228]: The preferences file cannot be opened or it is corrupted, so we will use default values
2021-04-23 17:00:37,524 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [236]: The preferences file 'None' cannot be found, will use default values
2021-04-23 17:00:37,524 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [239]: Completed loading preferences.
2021-04-23 17:00:37,531 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [274]: Created system tray icon.
2021-04-23 17:00:38,524 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 0.9898872375488281 seconds
```

guate6, 2021-04-24:
Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

guate6, 2021-04-24:
Can't upload them, but here's the links
https://www.dropbox.com/s/3lf4tg0sxsa0sit/cura%20%282%29.log?dl=0
https://www.dropbox.com/s/4rwkovmut4w57ae/cura%204.8.log?dl=0
https://www.dropbox.com/s/gcii7a67gscfzmf/cura.log?dl=0

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

Ghostkeeper, 2021-05-07:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9502.

In the log file it looks like it's not using tree support. However normal support is also generated single-threadedly in order from top to bottom (this just happens to be quite fast and much less significant than with tree support).

closed 2021-05-07 by Ghostkeeper

Ghostkeeper, 2021-05-11:
> Does you closing it mean you've identified the problem and will work to
resolve it?

We've identified that you're seeing the same issue as the other guy. We haven't been able to reproduce the issue yet though aside from the known issue with tree support (which is in our backlog). If we're able to reproduce the issue we'll determine priority with the team depending on how hard it is to fix (probably pretty hard in this case), how much it is obstructing the workflow (quite severely obstructing in this case) and how many people it affects (unknown at the moment, but it seems to be dependent on certain models from your description).

> How on earth would I use tree supports with this efficiently so all pieces
print out correctly?

By setting the Support Structure field to Tree, uptuning the collision resolution a bit and then having a lot of patience.