Ultimaker/Cura #9639: Fast heating

Matszwe02, 2021-04-21:
### Is your feature request related to a problem?

Cura as a default gcode uses M109 and M190, so the printer firstly homes, then heats up nozzle, then bed and then starts printing

### Describe the solution you'd like

The solution will be the following gcode:

```
M140 S65		;set heatbed to 65
M104 S210		;set hotend to 210
G28		;zero
M109 S200		;wait until hotend 200
M104 S205		;set hotend 205
M190 S60		;wait until bed 60
M140 S65		;set bed 65
```

### Describe alternatives you've considered

Of course everyone can write their own start gcodes, but this will be a great alternative to these who don't know as much gcode, but want to simply start faster.

### Affected users and/or printers

The update will work on every printer, there can be a checkbox in cura settings to replace standard home and heat up with my custom one. Printers with bed leveling will have a massive increase in preparation speed, because the leveling takes similar amount of time to the heating

GregValiant, 2021-04-21:
The Start-Up and End Gcodes come from the printer definition files.  The printer definition files are written by third parties usually contracted by the printer manufacturer (but sometimes someone from the community).  The manufacturer then submits the definition file to Ultimaker for inclusion in Cura.  For instance - my Ender definition was written by Trouch.com.  So Ultimaker and Cura don't have anything to do with it (except being nice enough that they include the definition in the program).
Your proposed start gcode doesn't mention G29.  When auto-leveling some people like to do different things with the temps to avoid drooling all over the build plate just before a print starts.  I personally hate to drool.  It makes me look even dumber than normal.

If you have the "Printer Settings" plugin loaded, you can de-select "Wait for build plate heat-up".  That speeds things up but your power supply has to be able to provide full heat-up power to the hot end and the bed at the same time.  Not all can do that.

fvrmr, 2021-04-22:
Hi @Matszwe02 thank you for your feature request.
@GregValiant explains this perfectly, so I will close this issue. 
Hope you understand!

closed 2021-04-22 by fvrmr

Ghostkeeper, 2021-05-07:
The default order for this is one that is more on the safe side, since some printers can't draw enough power to heat the bed and nozzle at the same time.