Ultimaker/Cura #9641: Marketplace

MrCurci, 2021-04-21:
### Application Version

4.9.0

### Platform

MacOS

### Printer

Two Trees Bluer

### Reproduction steps

Hi I have install the new version 4.9.0 and the Marketplace doesn't work.

### Actual results

So every time I try to use the Marketplace on Cura 4.9.0 there's no way I can access the Plugin and Material, the message shows on the marketplace box
"Could not connect to the Cura Package database. Please check your connection"
And if I try to go on Ultimaker Marketplace online, I select the Plugin and press to add to my Cura shows "Added to Cura"
but when I go to Cura 4.9.0 and press "Check for account updates" a triangular sign with Exclamation mark on the inside ⚠️ so doesn't do the update.

### Expected results

Work

### Additional information

Please Help , I would love to add some plugins to my Cura.

jellespijker, 2021-04-21:
Could you provide the log files?

Could you try login out of your account in a browser https://marketplace.ultimaker.com/app/cura/plugins 
and then Log out and in again in Cura.
[image]

MrCurci, 2021-04-21:
yep just done that , but still the same.

jellespijker, 2021-04-21:
Then we definitely need the log files.

KenLucke, 2021-04-21:
Just a data point... I do not see this issue on either the Mac or Windows version of 4.9.0 on my system... I have gone to the marketplace in both and upgraded some plugins without issue.

MrCurci, 2021-04-21:
Just don't know what's wrong with my Cura, I have uninstall and install again, made a new account but still no working

nallath, 2021-04-22:
We still need the log files. Please add the log files. We can't help you without them.

MrCurci, 2021-04-22:
Ok I kind solve my problem, what I've done was, I install Cura 4.9.0 on my wife's laptop download the plugin that I was looking for, clicked on Help then Show Configuration Folder, open the plugin folder then copy all the plugins folders then send by airdrop to my laptop and paste those folders on the same location but o my laptop, look like everything is working as planned.

MrCurci, 2021-04-22:
@nallath where do I find the log files?

nallath, 2021-04-22:
> @nallath where do I find the log files?

https://github.com/ultimaker/cura/#logging-issues

MrCurci, 2021-04-22:
[cura.log](https://github.com/Ultimaker/Cura/files/6356644/cura.log)

nallath, 2021-04-22:
You're having an issue with the way you setup proxies. I've found a guide that shows you what settings you should check: https://help.topazlabs.com/hc/en-us/articles/360037459772-SOCKS-version-5-protocol-error

MrCurci, 2021-04-22:
@nallath  I've just got home and follow the guide , everything is working now, Thank you for been patient. 🎊🎉🎊🎉👏🏼👏🏼👏🏼

closed 2021-04-22 by nallath