Ultimaker/Cura #9642: double g code from 1 slice

Reddy3d, 2021-04-21:
### Is your feature request related to a problem?

double gcode from one slice,for pause print and reload with other color.

### Describe the solution you'd like

plane cutting at "x"layer and export 2 gcode

### Describe alternatives you've considered

simplify 3d have a function for repair exporting 2 gcode for repair printing failure.

### Affected users and/or printers

ender 3v2

fvrmr, 2021-04-22:
Hi @Reddy3d thank you for your request.
I think you might wanna check the post processing script pause at height. Because I think that is what you are looking for. 
I will close this issue, but feel free to reach out if you have questions or it doesn't work for you.

Reddy3d, 2021-04-22:
IngleseItalianoAfrikaansAfrikaansAlbaneseAmharicAraboArmenoAssameseAzero (alfabeto latino)BengaleseBosniacoBulgaroCantonese (tradizionale)CatalanoCecoCinese semplificatoCinese tradizionaleCoreanoCroatoDaneseDariEbraicoEstoneFarsiFigianoFilippinoFinlandeseFranceseFrench (Canada)GalleseGiapponeseGrecoGujaratiHaitianoHindiHmong DawIndonesianoIngleseInuktitutIrlandeseIslandeseItalianoKannadaKazacoKhmerKiswahiliKlingon (Latin)Kurdish (Central)Kurdish (Northern)LaoLettoneLituanoMalayalamMaleseMalgascioMalteseMaoriMarathiMyanmar (Burmese)NepaleseNorvegeseOdiaOlandesePashtoPolaccoPortoghese (Brasile)Portoghese (Portogallo)PunjabiQuerétaro OtomiRumenoRussoSamoanoSerbo (alfabeto cirillico)Serbo (alfabeto latino)SlovaccoSlovenoSpagnoloSvedeseTahitianoTailandeseTamilTedescoTeluguTigrinyaTonganoTurcoUcrainoUnghereseUrduVietnamitaYucatec MayaI know the pause,but I need to create 2 different gcodes.to stop the printing and resume it later.example:mold a white base clock (a gcode)the machine ends as a work end.numbers and notches of black time, it must start from scratch as a new job, but from the point where it stopped with the first gcode.
--

i'm going to set up a modified g code divided into two parts as an example

Reddy3d, 2021-04-22:
I know the pause, but I need two separate gcodes, I set up two sample files that I created

[p1.txt](https://github.com/Ultimaker/Cura/files/6360000/p1.txt)
[p2.txt](https://github.com/Ultimaker/Cura/files/6360001/p2.txt)

Ghostkeeper, 2021-05-07:
Is there a problem with clicking the "save to disk" button twice?

Reddy3d, 2021-05-08:
[gcode filament 1.txt](https://github.com/Ultimaker/Cura/files/6445225/gcode.filament.1.txt)
[gcode filament 2.txt](https://github.com/Ultimaker/Cura/files/6445226/gcode.filament.2.txt)
[image]
[image]
i need of this two gcode from  same file,with start position different and with start -end gcode

Ghostkeeper, 2021-05-10:
Yes, I get it.
And why does the existing feature of "Filament Change" not work for you then?

Reddy3d, 2021-05-10:
I enter the wire change script, but it doesn't create two gcodes, and I don't see any m600 command

Reddy3d, 2021-05-10:
[image]

[test.txt](https://github.com/Ultimaker/Cura/files/6453785/test.txt)
I hoisted the test file and the screen

fvrmr, 2021-05-11:
On rule 6199 I find a M600 command:
```
M600 E30.00 L300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:9
;TYPE:WALL-INNER
;MESH:bearing_v_ring.stl
```

Ghostkeeper, 2021-05-17:
Indeed, it then still doesn't create two g-code files. It creates one, with a break in between to allow you to change filaments to create multi-colour prints.

We see this as a better solution than to build a feature that automatically chops g-code files for you.

closed 2021-05-17 by Ghostkeeper