Ultimaker/Cura #9643: Bad display of floating point numbers

Honi60, 2021-04-21:
### Application Version

4.9.0

### Platform

Win 10

### Printer

mine build

### Reproduction steps

Just install the latest version 4.9.0

### Actual results

Bad visual of numbers see attach file

### Expected results

Normal numbers display

jellespijker, 2021-04-21:
It seems the attachment that you mentioned wasn't added. Can you add it again?

Honi60, 2021-04-23:
I send again the screen shot, see that numbers appears distorted

Hanina Golan

jellespijker, 2021-04-23:
Unfortunately, GitHub doesn't allow attachments to be sent via email. Could you add it via the browser?

jellespijker, 2021-04-23:
@Honi60 Could you take a look at issue #9667 There is a screenshot attached with that issue and I believe they are the same bug.

closed 2021-05-10 by no-response[bot]