Ultimaker/Cura #9644: Cura 4.9.0 Crash during Ender 3 selection

btpavao, 2021-04-21:
### Application Version

4.9.0

### Platform

W10 64

### Printer

Ender 3 

### Reproduction steps

After Cura 4.9.0 installation, at add my printer, right after selecting Ender 3 (offline), Cura crashes.  I've successfully added Ender 2 and 4, but Ender 3 and Ender 3 Pro crashes.
crashes.https://guides.github.com/features/mastering-markdown/

### Actual results

Cura closes, no printer added.

### Expected results

Add and comunicate with Ender 3

fieldOfView, 2021-04-22:
We really need your Cura.log to diagnose the issue. See https://github.com/ultimaker/cura/#logging-issues

btpavao, 2021-04-22:
Please, find attached the logs.
[cura.log](https://github.com/Ultimaker/Cura/files/6359653/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6359650/DxDiag.txt)

fieldOfView, 2021-04-22:
It could be that Cura does not like the unicode characters in your windows profile name. You might want to try creating another user without unicode characters, and see if Cura runs better when you are logged in as that user.

btpavao, 2021-04-22:
Thanks for the quick response! 

I'll try, though it works with exactly same user name with Cura 4.8.0 without any problems, also the issue seems to be only on Ender 3 profile, while other printers model and brend is ok. Currently I'm using both 4.8.0 and 4.9.0.

Ghostkeeper, 2021-07-21:
So far we haven't seen this issue ourselves, but the problem appears to be that the temp folder contains non-Latin characters. Since the temp folder is located in the user folder, it occurs when the Windows username contains non-Latin characters.

We currently have a ticket to solve this issue upstream in Scipy if we can within reasonable time. Developers, see issue CURA-8416.

Ghostkeeper, 2021-07-21:
A workaround may be to adjust the environment variable `%TEMP%` in the Windows environment variables configuration screen.

nallath, 2021-07-22:
4.11 will have a partial fix for this; It won't be fully fixed, but at least it shouldn't crash anymore.

fvrmr, 2021-09-10:
It seems like it isn't fixed in 4.11. I have made a ticket on our backlog for this. Developers see CURA-8552

A work around could be to change your path of the temp folder:

- Search for environment variables in the windows search bar. 
- This will open system properties and there you can click on Environment Variables. 
- Edit the TEMP variable. 
- Change the Variable value to something like C:\Temp

Here you can read more about modify environment variables: https://docs.oracle.com/en/database/oracle/machine-learning/oml4r/1.5.1/oread/creating-and-modifying-environment-variables-on-windows.html#GUID-DD6F9982-60D5-48F6-8270-A27EC53807D0

Goodfeat, 2022-05-26:
[image]
After installing cura 5, problems with opening files began, most of the .stl does not open
no matter where they are, locally or in storage, Latin or other letters. files still won't open

Goodfeat, 2022-05-26:
[image]
Can't open most stl files on сura 5

MariMakes, 2022-10-10:
We've made efforts to resolve this with https://github.com/Ultimaker/Cura/pull/13394 and https://github.com/Ultimaker/Uranium/pull/855

Closing this issue since it should be resolved for for Cura 5.2 Stable.
If it's not, please feel free to open the issue again and we'll take another look.

closed 2022-10-10 by MariMakes