Ultimaker/Cura #9645: Cura 4.9 crash

snoopyduy, 2021-04-22:
### Application Version

4.9

### Platform

Windows 7 64b

### Printer

Ultimaker S5

### Reproduction steps

App crashed at first start.

### Actual results

Can't run app

### Expected results

run app

fvrmr, 2021-04-22:
Hi @snoopyduy thank you for your bug report. 
Could you try to update described here: https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-676851508

Please let me know if that will solve your issue.

stolz55, 2021-04-24:
Installing Microsoft security upgrade KB3126587 worked for me. Win 7 64b.  I was getting the same error message when trying to start Cura 4.9.   Now it opens correctly.

fvrmr, 2021-04-29:
Good to know that the upgrade worked for you @stolz55

closed 2021-05-06 by no-response[bot]