Ultimaker/Cura #9646: Add operative system to Cura installation files amd64, Darwin, etc

printingotb, 2021-04-22:
### Is your feature request related to a problem?

When installing a beta you get to choose between 6 different downloads and while obvious to some, it can be difficult to know what file to use. This potentially reduces the amount of beta-testers. I have installed several betas and I still don't remember what file I used last time :grimacing: 
[image]

### Describe the solution you'd like

Make it easier to see what file I should download for my operating system.

### Describe alternatives you've considered

------

### Affected users and/or printers

every potential beta tester

PS: the additional information & file uploads section in this form is missing.

fvrmr, 2021-04-22:
Hi @printingotb thank you for your feature request. Will discuss it with the team.

And good one, I will update the forms 😄

fvrmr, 2021-04-26:
Hi @printingotb I have discussed it with the team. And we've decided to defer this even though it seems like a small change because we don't think it will help other users. Hope you understand.

fieldOfView, 2021-04-26:
> because we don't think it will help other users

For the record, I disagree with that sentiment.

fvrmr, 2021-04-29:
> For the record, I disagree with that sentiment.

That's okay. Discussing it in team we had different opinions as well. But we had to make a decision.

Ghostkeeper, 2021-05-07:
Wait, so was it deferred (i.e. it doesn't have enough priority for the amount of work) or was it rejected (i.e. it's not adding value)?

I can't imagine that changing the name from `Ultimaker_Cura-4.9.0-BETA-amd64.exe` to `Ultimaker_Cura-4.9.0-BETA-win64.exe` would be a lot of work.

fvrmr, 2021-05-07:
@Ghostkeeper no it wouldn't be a lot of work. In that case the label rejected is better. 

@printingotb that might also be a good solution, but since beta's are only on Github and not on the website it wouldn't make a difference.

fieldOfView, 2021-05-07:
I am getting more and more curious about reasons NOT to change this.

printingotb, 2021-06-09:
Not to be overly persistent :grimacing: , I trust that you make the optimal decision all things considered but this showed up on Facebook:
Making it as easy as possible to install betas doesn't seem like a bad idea.
[image]

nallath, 2021-06-09:
> Not to be overly persistent grimacing , I trust that you make the optimal decision all things considered but this showed up on Facebook:
> Making it as easy as possible to install betas doesn't seem like a bad idea.
> [image]

I guess we could do some things to make it easier, but not understanding how an installer works (or that an .exe file is something you can execute) isn't something that we should try and solve...

Ghostkeeper, 2021-06-17:
The website does make it easier, of course, with this nice unbalanced design:
[image]

But I agree that for beta testers who don't know about file extensions we could make things easier by changing the file names on Github.