Ultimaker/Cura #9647: Win10 Cura 4.9.0 error

hrf3, 2021-04-22:
### Application Version

4.9

### Platform

Win 10

### Printer

none

### Reproduction steps

Starting as user fails.

### Actual results

Starting as admin works.

### Expected results

Starting Cura.

jellespijker, 2021-04-22:
@fvrmr and other devs  Also mentioned on FB by two other people: see https://www.facebook.com/groups/1959877270949115/?multi_permalinks=2924113354525497

jellespijker, 2021-04-22:
@hrf3 could you add your log files, these would really help out.

MarkusvonStaden, 2021-04-22:
I have the same Issue. There are no log files. I only get the DxDiag Log:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6358248/DxDiag.txt)

hrf3, 2021-04-22:
The stderr.log, stdout.log and cura.log are empty.

nallath, 2021-04-22:
Do you have a X: drive mounted?

hrf3, 2021-04-22:
Sure, it's a disconnected online drive

chimcen, 2021-04-22:
Got the same error - have also a drive "x" mounted (network drive)...

fvrmr, 2021-04-22:
You could try to rename that drive. Please let us know if that will fix it.

MarkusvonStaden, 2021-04-22:
Yeah, that solved it for me.

bjhiltbrand, 2021-04-22:
Same issue here. Deleting my network mapped drive fixed it for me as well.

chimcen, 2021-04-23:
My Drive on letter "X" is a NAS drive. When my NAS is offline i got this error Message wich prevents Cura 4.9 (4.8 works without any problems) from starting.

When my NAS is online and the drive on "X" is accessible, Cura 4.9 starts also without problems.

Could you fix this please? It's annoying to ensure that the drive mapped on "X" is online before starting up Cura.
Mounting no Drive to "X" isn't an option...

Thank you :)

smartavionics, 2021-04-23:
Out of curiosity, could you please try my Cura build to see if it has the same problem as the UM build regarding drive X. Installing it won't conflict with any other installation. You can find my latest release at https://github.com/smartavionics/Cura/releases/tag/20210420. Thanks.

fieldOfView, 2021-04-23:
Could the Cura build server use an mapped folder instead of mapping a drive, so that there's no longer a driveletter baked into the binaries?
https://www.askvg.com/create-a-virtual-drive-for-your-desired-folder-in-my-computer-using-subst-command/

fieldOfView, 2021-04-23:
Sorry, I meant the other way around, ignore the link to SUBST.

Use an NTFS junction that maps the drive to a folder on C:. Then at least the baked in path points to a drive that can be reasonably expected to always we there.

LAP87, 2021-04-25:
Why is cura looking for X: drive at all?
I'm using X: for a mounted cloudstorage, and cura 4.9 crashes at boot with this message.
[image]

Ghostkeeper, 2021-05-07:
> Why is cura looking for X: drive at all?

We don't rightly know, but it seems to be some weirdness with how we use cx_Freeze, the application that packages the Python code into an executable. The X: drive is the drive that Cura is built on in our build server. So the path to the Python files is on the X: drive there, and that is the last time the Python files will ever be an actual file; later they will be packaged inside cura.exe. At that point the `__file__` attribute is frozen in-place. These file names on the build server are stored in the Python packages. At some point, something seems to try to access files relative to the `__file__` of these Python packages. It's not in our own code, or at least it wasn't as far as I could find about 2 years ago. But some of our libraries might still try this.

For Cura 4.9.1 we've moved this to the C: drive on that build server. This is going to be more robust since many applications already assume that Windows is installed in the C: drive anyway.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/9600.

closed 2021-05-07 by Ghostkeeper