Ultimaker/Cura #9648: Win 10 crash in QtCore5.dll (4.9.0)

pmarkiewicz, 2021-04-22:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Unrelated

### Reproduction steps

File open cause crash and cure closes without any error message
From windows log crash is in Qt5Core.dll

Details below
Nazwa aplikacji powodującej błąd: Cura.exe, wersja: 4.9.0.0, sygnatura czasowa: 0x6064b6cf
Nazwa modułu powodującego błąd: Qt5Core.dll, wersja: 5.15.2.0, sygnatura czasowa: 0x5fa4dd3b
Kod wyjątku: 0xc0000409
Przesunięcie błędu: 0x00000000000204e8
Identyfikator procesu powodującego błąd: 0x250
Godzina uruchomienia aplikacji powodującej błąd: 0x01d736cfd406586c
Ścieżka aplikacji powodującej błąd: C:\Program Files\Ultimaker Cura 4.9.0\Cura.exe
Ścieżka modułu powodującego błąd: C:\Program Files\Ultimaker Cura 4.9.0\lib\Qt5Core.dll

### Actual results

Application closewith any info (only information are in windows logs)

### Expected results

File select dialog is expected

chrisgaw175, 2021-04-22:
I have the same problem

pmarkiewicz, 2021-04-22:
You may try to unistall chitubox

chrisgaw175, 2021-04-22:
I've downgraded to 4.8 cura

fvrmr, 2021-04-22:
> You may try to unistall chitubox

Does Cura works now when you did this?

pmarkiewicz, 2021-04-23:
yes

chrisgaw175, 2021-04-23:
ok, but I also need chitubox, and cura 4.8 IS working aside with chitubox, so UM changed something which is in conflict with other software, it can also be in conflict for example with CAD's there can be conflict which effects are hard to notice like sometimes making a not full layer (one need to slice again to correct the missing part of outer wall in one layer)

fvrmr, 2021-04-23:
I am not able to reproduce this. What version of Chitubox are you using?

pmarkiewicz, 2021-04-23:
no idea, just removed one, probably it was 6-9 months old

Liger0, 2021-04-26:
maybe my ticket is related then #9674 
I have chitubox x64 v1.8.1
when installed I selected to enable stl thumbnails.
It may explain why, when going through directories to find stl files, cura 4.9 crashes when you enter a folder that contains an stl.
Cura 4.8 works fine.

fvrmr, 2021-04-26:
@Liger0 thank you. I've tried that to. But I still can't reproduce this unfortunately. When you disabled that settings Cura works for you?

Liger0, 2021-04-26:
> 
> 
> @Liger0 thank you. I've tried that to. But I still can't reproduce this unfortunately. When you disabled that settings Cura works for you?

You can't disable that setting unfortunately.

Liger0, 2021-04-27:
> 
> 
> @Liger0 thank you. I've tried that to. But I still can't reproduce this unfortunately. When you disabled that settings Cura works for you?

I gave the command `regsvr32 /u "CHITUBOX_Thumbnail.dll"`
and, after restarting the OS, cura stopped crashing when opening folders.
The dll file is still in the chitubox folder, but it doesn't work anymore.

fvrmr, 2021-04-29:
I will close this issue since it is a duplicate: #9668 
There you can find a workaround and an explanation about this issue.

Please continue the thread there, so we keep everything on one place.

closed 2021-04-29 by fvrmr