Ultimaker/Cura #9649: Plantage Cura 4.9 & OSX.13.6

orkide11, 2021-04-22:
### Application Version

4.9

### Platform

OSX.13.6

### Printer

Artillery Sidewinder X1

### Reproduction steps

Le logiciel plante systématiquement

### Actual results

Le logiciel plante systématiquement

### Expected results

Le logiciel plante systématiquement

fieldOfView, 2021-04-22:
If the actual results and the expected results match, then what's the issue? Please fill in the issue template in english, and provide details and preferably logs, or there is nothing that we can do to diagnose and fix the issue you have.

jwrw, 2021-04-22:
Noting the O/S version - potentially related to #9631 ?

fvrmr, 2021-04-23:
I will close this issue since @orkide11 commented on that issue.
So this issue is a duplicate of #9631

closed 2021-04-23 by fvrmr