Ultimaker/Cura #9650: Downgrade to 4.9

PhilSadler57, 2021-04-22:
### Application Version

4.9

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Opened app. drag and drop file for printing (as previously worked) - CRASH
2. Opened app - tried open file from file menu - CRASH
3. Repeat with other files - all of which worked fine previously.
All was fine before - I am looking for the previous (WORKING) version and re-install....

### Actual results

CRASH, CRASH, CRASH...

### Expected results

Simply - work... as it used to

### Additional information

.

fieldOfView, 2021-04-22:
We can't help you with the crashes unless you provide logs. Without logs I can only guess what is going on. Do you perhaps have the "Printjob Naming" plugin installed? If so, please check the Installed tab of the Marketplace for an updated version which fixes an incompatibility with Cura 4.9. It was published to the Marketplace a month ago.

Previous versions of Cura can be found here: https://github.com/Ultimaker/Cura/releases
Specifically version 4.8 is here: https://github.com/Ultimaker/Cura/releases/tag/4.8.0

PhilSadler57, 2021-04-22:
I am just a simple user, I have no idea how to seek and provide logs of which you speak.

 

I have reverted to 4.8 as it simply just works.

 

I try keep things largely vanilla, the only plug-ins I use being;

 

Keep well

 

 

phil

Asterchades, 2021-04-22:
> I am just a simple user, I have no idea how to seek and provide logs of which you speak.
The form presented when you posted your issue provides instructions on how to provide a log. It's between the "Additional information & file uploads" header and the "Additional information" section, along with the requests for project files (explaining how to create them) and screenshots. It specifically links to [here](https://github.com/Ultimaker/Cura#logging-issues).

closed 2021-05-06 by no-response[bot]