Ultimaker/Cura #9651: Supports and Support Blockers

cwschwartz01, 2021-04-22:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Tronxy SA PRO

### Reproduction steps

Used supports and it reduced the print volume so much it was unusable. The same this happened when I use a support blocker. I am using CURA 4.9.

### Actual results

The Print volume was reduced so small that I could not use support blockers or supports.

### Expected results

Small reduction in print volume. 

### Additional information

None

fieldOfView, 2021-04-22:
We can't even begin to know what went wrong without a better description of what you did, one or more screenshots, or better yet a project (File -> Save project...).

cwschwartz01, 2021-04-22:
Before 

[image]

After Supports added

[image]

Add Support Blocks

[image]

GregValiant, 2021-04-22:
Special Modes | Print Sequence is set to "One at a Time" and so your MaxZ is determined by your gantry height.  Change the setting to "All at Once".

fvrmr, 2021-04-23:
I will close this issue since it is solved!

closed 2021-04-23 by fvrmr

Ghostkeeper, 2021-05-10:
Arguably, a support mesh should not be seen as a separate mesh in one-at-a-time mode. This is being tracked in https://github.com/Ultimaker/Cura/issues/4193.